gnuradio/gnuradio #1457: SEGFAULT on macOS 10.12

gmichael225, 2017-09-01:
Apologies if I'm missing something obvious, but I've installed the gnuradio-devel and uhd-devel ports as suggested in other issues where this has come up, but I'm still getting a segmentation fault when launching gnuradio-companion.

`Process:               Python [3193]
Path:                  /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.13 (2.7.13)
Code Type:             X86-64 (Native)
Parent Process:        zsh [15642]
Responsible:           Python [3193]
User ID:               501

Date/Time:             2017-09-01 16:07:36.881 +1000
OS Version:            Mac OS X 10.12.6 (16G29)
Report Version:        12
Anonymous UUID:        045588E5-A464-2FE3-2BF5-EBDA6590B1AB

Sleep/Wake UUID:       6B1381D0-29AE-45DD-86D3-17D1B8970EEB

Time Awake Since Boot: 39000 seconds
Time Since Wake:       6100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0:
--> 
    __TEXT                 000000010e9ed000-000000010e9ef000 [    8K] r-x/rwx SM=COW  /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_c.dylib             	0x00007fffdc988b52 strlen + 18
1   org.python.python             	0x000000010ea4e847 PyString_FromString + 23
2   org.python.python             	0x000000010ea5319d PyString_InternFromString + 13
3   org.python.python             	0x000000010ea47216 PyObject_SetAttrString + 54
4   _gobject.so                   	0x000000011244afd0 pygobject_new_with_interfaces + 564
5   _gobject.so                   	0x000000011244b0a2 pygobject_lookup_class + 108
6   _gobject.so                   	0x000000011244ab7d pyg_type_get_bases + 108
7   _gobject.so                   	0x000000011244a921 pygobject_register_class + 74
8   _gio.so                       	0x00000001136459dc pygio_register_classes + 1595
9   _gio.so                       	0x00000001136415e1 init_gio + 533
10  org.python.python             	0x000000010eacb2ec _PyImport_LoadDynamicModule + 140
11  org.python.python             	0x000000010eac9cef import_submodule + 271
12  org.python.python             	0x000000010eac9778 load_next + 280
13  org.python.python             	0x000000010eac87b2 PyImport_ImportModuleLevel + 1138
14  org.python.python             	0x000000010eaa17c7 builtin___import__ + 135
15  org.python.python             	0x000000010ea03203 PyObject_Call + 99
16  org.python.python             	0x000000010eaac429 PyEval_EvalFrameEx + 22425
17  org.python.python             	0x000000010eaa6947 PyEval_EvalCodeEx + 2119
18  org.python.python             	0x000000010eaa60f6 PyEval_EvalCode + 54
19  org.python.python             	0x000000010eac71a1 PyImport_ExecCodeModuleEx + 241
20  org.python.python             	0x000000010eaca273 load_source_module + 1059
21  org.python.python             	0x000000010eaca5ef load_package + 303
22  org.python.python             	0x000000010eac9cef import_submodule + 271
23  org.python.python             	0x000000010eac9778 load_next + 280
24  org.python.python             	0x000000010eac87b2 PyImport_ImportModuleLevel + 1138
25  org.python.python             	0x000000010eaa17c7 builtin___import__ + 135
26  org.python.python             	0x000000010ea03203 PyObject_Call + 99
27  org.python.python             	0x000000010ea03366 PyObject_CallFunction + 246
28  org.python.python             	0x000000010eac8202 PyImport_Import + 386
29  org.python.python             	0x000000010eac661f PyImport_ImportModule + 31
30  _gtk.so                       	0x000000011247e324 pygtk_register_classes + 29
31  _gtk.so                       	0x000000011246a628 init_gtk + 818
[... 370 lines cut ...]
__LINKEDIT                       124.1M      104 
__TEXT                           178.0M      302 
__UNICODE                          556K        2 
shared memory                       12K        4 
===========                     =======  ======= 
TOTAL                            470.4M      767 

Model: MacBookPro13,3, BootROM MBP133.0226.B25, 4 processors, Intel Core i7, 2.9 GHz, 16 GB, SMC 2.38f7
Graphics: Radeon Pro 460, AMD Radeon Pro 460, PCIe, 4096 MB
Graphics: Intel HD Graphics 530, Intel HD Graphics 530, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445424533303445422D45474347202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x15A), Broadcom BCM43xx 1.0 (7.21.171.130.1a1)
Bluetooth: Version 5.0.5f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Belkin USB-C LAN, Ethernet, en7
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: iBridge
USB Device: USB2.0 Hub
USB Device: USB Receiver
USB Device: USB3.0 Hub
USB Device: USB Storage
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1
`

vamsi765, 2017-09-01:
I think the issue you have is with gnuradio. My issue was with uhd.

I think your issue is with python packages. Probably multiple copies of gnuradio.

gmichael225, 2017-09-05:
No multiple copies so far as I can tell!

michaelld, 2020-02-27:
Is this still an issue, some 2.5 years down the road? I have no issue running GRC from GR37 or GR38 on my OSX 10.14 or 10.15 boxes.

nickoe, 2021-01-03:
We better close this one.

closed 2021-01-03 by nickoe