gnuradio/gnuradio #1459: integrate has `int vlen` parameter; should be `unsigned`; check other block's

marcusmueller, 2017-09-05:
There's a lot of blocks that take `vlen` parameters. This is a reminder to myself (please assign to me) to go through the tree and identify all places where we should've used unsigned int instead of int.

This might look cosmetic (it's unlikely someone would want a vector half of MAX_INT in size, and the scheduler and buffer allocator might reasonably break on that), but it actually gives compiler warnings when trying to parameterize a block with a signed vlen, so it's actually a code quality issue.

vamsi765, 2017-10-07:
@marcusmueller Do you want to me to take a look at the issue?

nishithshah2211, 2017-10-26:
@marcusmueller Does this issue still need work? Some more info to identify such occurrences of `int` usage instead of `unsigned int` would also help.

closed 2018-11-01 by marcusmueller