gnuradio/gnuradio #1460: 403 Forbidden nginx - https://gnuradio.org/

saydulk, 2017-09-08:
showing 403 Forbidden nginx at gnuradio.org website, please fix nginx 
check - service nginx status or ps waux | grep nginx

BrashEndeavours, 2017-09-08:
Works for me.

saydulk, 2017-09-08:
[image]

marcusmueller, 2017-09-08:
Works for me, too, so, I'd assume you, @saydulk, you might be 

* behind a misconfigured proxy or
* you might be blocked.

Either case can't be solved via the source code issue tracker. I'm recommending closing this issue. 

If you are being blocked, we, or our CDN, might have been seeing attacks or Spam coming from your general direction. @noc0lour might have more insight on how to assess that, please contact him on IRC via freenode in #gnuradio .

closed 2017-09-08 by saydulk