gnuradio/gnuradio #1466: python: replace star imports in python modules

noc0lour, 2017-09-17:
In some submodules (e.g. analog) `__init__.py` has a lot of star imports to submodules which makes tracing the called function hard. Clashing function names in the submodule can also occur without noticing it.

Replacing star imports with named imports in `__init__.py` would make clean which functions are intended to be in the top-level package.

mbr0wn, 2017-09-26:
We should, in general, aim for a high PyLint score. It would flag star imports. Could we somehow integrate this into BuildBot?

noc0lour, 2017-09-26:
```
Raw metrics
-----------

+----------+-------+------+---------+-----------+
|type      |number |%     |previous |difference |
+==========+=======+======+=========+===========+
|code      |123787 |44.38 |NC       |NC         |
+----------+-------+------+---------+-----------+
|docstring |63092  |22.62 |NC       |NC         |
+----------+-------+------+---------+-----------+
|comment   |5975   |2.14  |NC       |NC         |
+----------+-------+------+---------+-----------+
|empty     |86058  |30.85 |NC       |NC         |
+----------+-------+------+---------+-----------+

Duplication
-----------

+-------------------------+------+---------+-----------+
|                         |now   |previous |difference |
+=========================+======+=========+===========+
|nb duplicated lines      |11326 |NC       |NC         |
+-------------------------+------+---------+-----------+
|percent duplicated lines |4.064 |NC       |NC         |
+-------------------------+------+---------+-----------+

Messages by category
--------------------

+-----------+-------+---------+-----------+
|type       |number |previous |difference |
+===========+=======+=========+===========+
|convention |24826  |NC       |NC         |
+-----------+-------+---------+-----------+
|refactor   |2262   |NC       |NC         |
+-----------+-------+---------+-----------+
|warning    |12576  |NC       |NC         |
+-----------+-------+---------+-----------+
|error      |45434  |NC       |NC         |
+-----------+-------+---------+-----------+

% errors / warnings by module
-----------------------------

+-------------------------------------------------+------+--------+---------+-----------+
|module                                           |error |warning |refactor |convention |
+=================================================+======+========+=========+===========+
|gnuradio.filter.filter_swig                      |7.98  |1.69    |5.17     |4.25       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.dtv.dtv_swig                            |7.83  |1.69    |8.58     |4.03       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.blocks.blocks_swig3                     |7.03  |1.53    |5.08     |2.80       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.blocks.blocks_swig2                     |6.74  |1.50    |5.44     |2.34       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.analog.analog_swig                      |6.71  |1.65    |3.98     |2.56       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.trellis.trellis_swig1                   |6.69  |1.30    |6.90     |9.85       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.blocks.blocks_swig5                     |6.21  |1.64    |5.08     |2.51       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.digital.digital_swig0                   |5.41  |1.65    |4.95     |3.98       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.blocks.blocks_swig1                     |5.33  |1.15    |3.49     |2.18       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.blocks.blocks_swig4                     |4.82  |1.13    |3.45     |1.89       |
+-------------------------------------------------+------+--------+---------+-----------+
|gnuradio.trellis.trellis_swig0                   |4.74  |1.15    |4.91     |4.80       |
+-------------------------------------------------+------+--------+---------+-----------+
[... 613 lines cut ...]
|useless-else-on-loop            |1           |
+--------------------------------+------------+
|unused-format-string-argument   |1           |
+--------------------------------+------------+
|too-many-boolean-expressions    |1           |
+--------------------------------+------------+
|redundant-unittest-assert       |1           |
+--------------------------------+------------+
|no-init                         |1           |
+--------------------------------+------------+
|format-combined-specification   |1           |
+--------------------------------+------------+
|expression-not-assigned         |1           |
+--------------------------------+------------+
|empty-docstring                 |1           |
+--------------------------------+------------+
|cell-var-from-loop              |1           |
+--------------------------------+------------+

Global evaluation
-----------------
Your code has been rated at -12.93/10
```

I don't think we want to check the swig generated parts, but yes integrating in buildbot is a good idea

sanchit97, 2017-09-29:
Hi! I am interested in doing this task. Can you point out any explicit examples where I can learn a little about the task?

balister, 2017-09-29:
I ran 
$ grep "import \*" -r *
and got a bunch of hits. See if that can get you started.

marcusmueller, 2017-10-07:
@sanchit97 checking back on you: Because you said you'd take on this task, we'd be willing to help you. So, how's it going? Is there any problem that you need help with?

sanchit97, 2017-10-07:
@marcusmueller Thanks for checking in. I tried finding all the star imports in the whole repo. The occurrences of such imports ran >150 in number. So, I was wondering if all of them have to be manually checked and changed?
Another thing is if encountering something like "from numpy import *". Does this need to be changed or not?

marcusmueller, 2017-10-07:
I'd say, yes.

By the way, `from numpy import *` is a prime example of something that shouldn't be done with a star import – for example, because `numpy.sum` then overrides the python built-in `sum`, which is a recipe for bugs.

Note that nobody's expecting you to get completely rid of all star imports at once. Start with a branch that gets rid of all star imports in `gnuradio-runtime`, for example. Or, find all `__init__.py`, and get rid all star imports that are not `from xyz_swig import *`. Something like that – make a start, don't stress yourself too much. Better do two pull requests than one, in case we have feedback for the first one that changes how you'd do the second.

Anyway, happy hacking!

noc0lour, 2017-10-08:
Heh, I didn't have a specific proposal for improving the code here. I was tracking down some issue and got annoyed by looking through modules which might import this specific function/class. I like the `from foo import bar, baz` syntax.

```
In [1]: from gnuradio import digital

In [2]: from inspect import getmodule

In [3]: for fun, point in digital.__dict__.iteritems():
   ...:     print("{}: {}".format(fun, getmodule(point)))
```

lakshmanaram, 2017-10-10:
Hey! If the work hasn't been started on this issue or isn't progressing, I'd like to take it up. 
Please confirm @sanchit97

sanchit97, 2017-10-10:
@lakshmanaram I started some work, but feel free to go ahead.

nishithshah2211, 2017-10-26:
Hey! If the work hasn't been started on this issue or isn't progressing, I'd like to take it up.
Please confirm @lakshmanaram

neilsh, 2017-10-31:
looks like somebody has a PR fixing a handful of files: https://github.com/gnuradio/gnuradio/pull/1491

marcusmueller, 2021-09-23:
this being 4 years old: @noc0lour could you re-run your analysis?

marcusmueller, 2023-12-21:
Ignoring all `__init__.py`, all doxygen-generated code:

```
gr-blocks/grc/blocks_ctrlport_viewer.block.yml
12:    imports: from gnuradio.ctrlport.monitor import *

gr-blocks/grc/blocks_ctrlport_performance.block.yml
12:    imports: from gnuradio.ctrlport.monitor import *

gr-filter/python/filter/design/filter_design.py
63:    from gnuradio.filter.banditems import *
69:    from gnuradio.filter.polezero_plot import *
76:#     from gnuradio.filter.idealbanditems import *
81:    from gnuradio.filter.api_object import *
87:    from gnuradio.filter.fir_design import *

gr-dtv/python/dtv/atsc_rx.py
23:from .atsc_rx_filter import *

grc/gui/Constants.py
12:from ..core.Constants import *

gnuradio-runtime/python/gnuradio/ctrlport/gr-ctrlport-monitor
20:from gnuradio.ctrlport.GrDataPlotter import *

gnuradio-runtime/python/gnuradio/ctrlport/gr-perf-monitorx
48:from gnuradio.ctrlport.GrDataPlotter import *
```

so, yeah, this persists, but at lower priority