gnuradio/gnuradio #1470: Compiling with gcc6/gcc7

rvl, 2017-09-25:
Hi, this is probably related to #1316.

Currently the Arch Linux and Fedora distributions patch gnuradio (e.g. [Arch](https://git.archlinux.org/svntogit/community.git/tree/repos/community-x86_64/gnuradio-gcc7.patch?h=packages/gnuradio&id=d61c3e4db7741753208286f32d33dbf26cd2bb53), [Fedora](http://pkgs.fedoraproject.org/cgit/rpms/gnuradio.git/commit/gnuradio-3.7.11-gcc-7-compile-fix.patch?id=b2f5f351c1181867ea28146e2cfd91d9d6d57cb4)).

I am about to do the same for Nix (see NixOS/nixpkgs#29740, gnuradio 3.7.11). In Nix, we are even still on gcc6 as the default compiler.

It would be really good if the next release included this patch, or an equivalent fix.

Cheers!

noc0lour, 2017-09-25:
Thanks for the report @rvl. #1424 and #1425 were merged in master and maint respectively.
Not sure about the `<memory>` includes you put in there. 
Could you provide a build log with the errors? Apparently the build logs from the failed builds in https://github.com/NixOS/nixpkgs/pull/29740 are not accessible anymore.

rvl, 2017-09-25:
Hi @noc0lour thanks for the info.

The failing build log is here: https://hydra.nixos.org/build/61662739/nixlog/1

It wasn't my patch -- I just grabbed it from Arch. But you're right, it appears those `<memory>` includes aren't necessary.

If the two PRs referenced fix builds with gcc6--gcc7 and boost 1.62--1.65 then we are happy, because then there will be no source patches or cflags required to package gnuradio.

closed 2017-09-27 by rvl

noc0lour, 2017-09-27:
@rvl so I guess the fixes on current master/maint work for you?