gnuradio/gnuradio #1474: uhd_fft

Praetorian72, 2017-09-28:
Good day to every body, 
I'm using an ETTUS B200 mini for doppler frequency monitoring.
To analyze the spectrum in real time, I'm currently using uhd_fft.py application whose maximum fft sample number is 4096, while I need 2^14 samples. 
I tried to investigate into the c++ files (invoked by uhd_fft.py), such as freq_sink_c_impl.cc (https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/lib/freq_sink_c_impl.cc) whose line 216 describes the following action:
216 throw std::runtime_error("freq_sink: FFT size must be > 16 and < 16384.");
The problem seems to be hard without the whole file list to be recompiled....
Would anybody help me ?
Thank you very much!!

mbr0wn, 2017-10-02:
@Praetorian72 It's not quite clear what your request is. Maybe `uhd_fft` is not the right tool for this kind of application. If you *do* want the QT Frequency Sink to have longer FFTs, recompilation is not optional.

Consider making your own application for this. The regular FFT block might be better suited, and the QT GUI Vector Sink could also be useful.

Praetorian72, 2017-10-03:
@mbr0wn  thank you very much for your replay. I believed it was possible, by recompiling, to increase fft sample number from 2^12 samples to 2^16 samples. Anyway, I will use the Vector Sink module as you suggested me. Further, I'm using also alternative (and surely more suitable) techniques implemented on specest_gui ..,,but on Fedora 23 I'm having some problems. Thank you again for your support. Best Regards, Frank

closed 2018-02-04 by mbr0wn

marcusmueller, 2018-02-11:
This is a general reminder:

The original question reads as if it is not a *bug report*. Notice that this is a *bug tracker*. The [discuss-gnuradio mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio) is the proper place to discuss approaches to solving complex problems. 

I'm really happy about every bug being reported here, even if it not turns out to be a bug. I just get this feeling that people in the future will use a search engine, find this page and comment on this not-really-a-bug-report with more discussion: Please don't. Use the mailing list for discussions! It's much more active.