gnuradio/gnuradio #1478: Unresolved conflicts in the homebrew file

chevdor, 2017-10-04:
It looks like someone committed with unresolved conflicts:
`/usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/gnuradio.rb` contains many conflicts. 
For instance:
```
<<<<<<< Updated upstream
  homepage "https://gnuradio.org/"
  url "https://gnuradio.org/releases/gnuradio/gnuradio-3.7.11.tar.gz"
  sha256 "87d9ba3183858efdbb237add3f9de40f7d65f25e16904a9bc8d764a7287252d4"
  revision 1
  head "https://github.com/gnuradio/gnuradio.git"
=======
  homepage "https://gnuradio.squarespace.com/"
  revision 2
  url "https://github.com/gnuradio/gnuradio.git",
      :revision => "dd3e3c8198515e10cc91145a16b08d0f37ee1bd1"
  version "4.0-alpha1"
  head "https://github.com/gnuradio/gnuradio.git", :branch => "next"
>>>>>>> Stashed changes
```

jmcorgan, 2017-10-04:
@michaelld ping

chevdor, 2017-10-04:
I guess the issue is local on my end as I don´t see the issue in https://github.com/Homebrew/homebrew-core/blob/master/Formula/gnuradio.rb

closed 2017-10-04 by chevdor

michaelld, 2017-10-04:
Thank goodness! I'm not particularly familiar with 'brew; I'm not sure how much help I could be!