gnuradio/gnuradio #1480: Error importing GNU Radio's wxgui

letitbezh, 2017-10-06:
when i use osmocom_fft  it is wrong
```shell
founder@founder-pc:~/hackrf/gr-drm/apps/gui$ osmocom_fft 
Error importing GNU Radio's wxgui. Please make sure gr-wxgui is installed.
```
the gnuradio is right .whar should i do ,thank you for your answer!

marcusmueller, 2017-10-07:
This is not the home of gr-osmosdr, from which `osmocom_fft` is. Also, `osmocom_fft` is not in gr-drm, and the obvious question here is whether you've got gr-wxgui installed. So, wrong place to ask!

You can check whether a GNU Radio core component is installed by running `gnuradio-config-info --enabled-components`. 

Note that gr-wxgui is deprecated and will be removed in the upcoming GNU Radio 3.8; when designing applications, you must avoid it.

kabzonkill, 2019-07-25:
how to install gr-wxgui

michaelld, 2019-07-25:
You have to use GR 3.7 to get gr-wxgui; it has been removed from all GR 3.8.

How to install it depends on your OS as much as anything. There are good install guides for GR that should help, such as < https://wiki.gnuradio.org/index.php/InstallingGR >.

kabzonkill, 2019-07-25:
but gr-wxgui is already in the gnuradio directory. The question is how to install it? I put the XML files in / usr / local / share / gnuradio / grc / blocks, and they appeared in the blocks, but not compiled (((

michaelld, 2019-07-25:
Is GNU Radio installed?
Can you execute `gnuradio-config-info --enabled-components`?
Does "wx-gui" appear in the list of enabled components?
If so, then it was built and installed; you just need to set various environment variables to get access to the Python and GRC scripts.
If not, then you need to go back to the CMake configuration step to see why gr-wxgui wasn't enabled & then figure out what needs to be done to enable it.

kabzonkill, 2019-07-25:
scored gnuradio-config-info --enabled-components!
no wx-gui!
How to install separately wx-gui do not tell me ???
I copied files from / usr / local / src / gnuradio / gr-wxgui / grc files
 xml in
/ usr / local / share / gnuradio / grc / blocks
and they appeared but did not work))
then i copy the python scripts to /usr/lib/python2.7/dist-packages/gnuradio/wxgui
of
/ usr / local / src / gnuradio / gr-wxgui / python / wxgui
it seems to compile, but the error produces no module: -
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import __ (name)
ImportError: No module named _wxgui_swig

michaelld, 2019-07-25:
You need the primary library (`/usr/local/lib/libgnuradio-wxgui.so`) and SWIG Python interface library (`_wxgui_swig.so`). Those come about from building the source or installing as a pre-built binary. You can't build them outside the main GR source.

michaelld, 2019-07-25:
Question: How did you install GR in the first place?

kabzonkill, 2019-07-26:
set GR_GSM and gr was included (

kabzonkill, 2019-07-26:
libraries found, copied! How to connect now?

kabzonkill, 2019-07-26:
now writes (( 
: - 
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/wxgui/__init__.py", line 29, in <module>
    dirname, filename = os.path.split(os.path.abspath(__file__))
NameError: name 'os' is not defined

kabzonkill, 2019-07-26:
include so

File "/usr/local/lib/python2.7/dist-packages/gnuradio/wxgui/wxgui_swig.py", line 22, in swig_import_helper
    _mod = imp.load_module('_wxgui_swig', fp, pathname, description)
ImportError: /usr/local/lib/python2.7/dist-packages/gnuradio/wxgui/_wxgui_swig.so: wrong ELF class: ELFCLASS32

dkozel, 2019-07-27:
Hi @kabzonkill. I sounds like you have a problem specific to your setup or application. Those are much better suited to being asked on the mailing list or IRC/Slack channel, not here on the development issue tracker.

https://wiki.gnuradio.org/index.php/MailingLists
https://wiki.gnuradio.org/index.php/IRC

We need as much information about your situation as possible to have any chance of helping. Please read this page and then ask your question on the Mailing List or IRC.
https://wiki.gnuradio.org/index.php/ReportingErrors#Checklist:_Asking_a_question_on_the_mailing_list

closed 2019-07-27 by dkozel