gnuradio/gnuradio #1483: runtime: d_min_noutput_items does not appear to do anything

mrjacobagilbert, 2017-10-12:
Setting this should constrain the number of output items requested to be produced by blocks according to the documentation, but it does not appear to do anything. Should this be:

- fixed such that it works as expected?
- left for future implementation and documentation updated?
- removed altogether?

jmcorgan, 2017-10-13:
The getter is defined inline in block.h and is called by the runtime in:

https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/block_executor.cc#L218

The value ultimately gets used by the logic in min_available_space(...):

https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/block_executor.cc#L79

Can you elaborate on the original issue?

mrjacobagilbert, 2017-10-13:
Interesting. This came out of discussion surrounding PR https://github.com/gnuradio/gnuradio/pull/1477 and I wrote a quick block that set this value to 1024 and expected that work() would not get called for values less than this, however it does.

I may be misunderstanding what this does, but ultimately I was not able to get flowgraph behavior to change based on what this value was set to.

mrjacobagilbert, 2017-10-28:
@jmcorgan - is there anything else to do here? If my understanding of the effect this parameter should have on FG execution is incorrect, this issue can just be closed.

@777arc - did you look into this further at all, or can you make sense of what we saw vs Johnathan said above?

777arc, 2017-10-28:
Well, it looks like I was wrong about the parameter "not being used", Johnathan shows how it is used in gnuradio-runtime/lib/block_executor.cc (it's clear that it's involved somehow in the computation of noutput_items).  That being said, when I try set_min_noutput_items(10000) I still get the typical 4096 items coming out.  So either my minimal example is incorrect in some way, or there is a bug.  I don't think we are misinterpreting what min_noutput_item is suppose to do, it seems fairly clear based on the name. 

I'm fine with closing the issue, just because it's less of an issue and more of a curiosity at this point.  Like, it's not preventing me from doing something I need to do.

mrjacobagilbert, 2017-10-28:
Likewise, I have tried set_min_noutput_items(1024) and work() gets called w/ 511 or 512 sometimes.

Its not limiting me either, and I can't think of a use for this that could not be addressed via forecast() anyway...but yeah, little odd.

giladbeeri, 2017-10-29:
It is a problem for me. I do need to get a minimal amount of output items,
and at the moment I'm using set_output_multiple() for that.

mrjacobagilbert, 2020-10-16:
I have done some additional work on this lately. I'm closing this issue as it does a poor job of explaining the issue and will open a new issue with slightly different/larger scope.

closed 2020-10-16 by mrjacobagilbert