gnuradio/gnuradio #1501: zeromq: any ZMQ source crashes on ARMv7/Ettus E310

noc0lour, 2017-10-30:
If I try to add any ZMQ source (REQ,SUB,PULL) I get this error accompanied with `Bus error` printed on stderr.

```
[  985.290186] Alignment trap: not handling instruction edc30b00 at [<b212e6b0>]
[  985.297239] Unhandled fault: alignment exception (0x811) at 0x00135a89
[ 1111.191859] Alignment trap: python2 (1482) PC=0xb20de6b0 Instr=0xedc30b00 Address=0x000a4891 FSR 0x811
```
This is with the tip of `master`.
Any clues?

devnulling, 2017-12-27:
Are you still running into this error? I cross compiled UHD 3.10.2.0 with the current GR master  - https://github.com/gnuradio/gnuradio/commit/a0adcd3347c7ffd6ef3c42ce7705a23978774d3b  using the release-4 SDK. The Sinks and Sources except for the PUB Sink worked as expected. I'm not seeing any error with the PUB Sink, there is just no data flow. I don't suspect the issue I'm seeing with the PUB Sink is due to any recent change as my flowgraph behaves the same with GR 3.7.9 (default release-4 GR version).
[image]

[image]

[image]

noc0lour, 2018-01-14:
Just retested with the E310 and the issue is gone. Though the kernel/FS is different now.

closed 2018-01-14 by noc0lour