gnuradio/gnuradio #1510: GRC Fails on Startup with a Memory Error

giladbeeri, 2017-11-03:
Using a pybombs prefix installed under `/opt/gr/stable` (installed the `gnuradio-stable` recipe), Ubuntu 16.04. Fresh installation.

When running `gnuradio-companion` from the CLI, it crashes and gives the error (in console):

```
user@comp /opt> gnuradio-companion
<<< Welcome to GNU Radio Companion 3.7.11.1 >>>

Block paths:
	/home/user/.grc_gnuradio
	/opt/gr/stable/share/gnuradio/grc/blocks
Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')fish: “gnuradio-companion” terminated by signal SIGSEGV (Address boundary error)

```

mbr0wn, 2017-11-10:
Not sure if calling this a memory error isn't a red herring. My gut still says abi mismatch.

giladbeeri, 2017-11-16:
How do I debug that?

mbr0wn, 2018-12-19:
Unclear what's going on, and not seen elsewhere. @giladbeeri please reopen if this is still an issue.

closed 2018-12-19 by mbr0wn