gnuradio/gnuradio #1511: Overlapping blocks and wires bug

abaxi, 2017-11-03:
Often times when one adds a block to the flow graph in GRC, it'll go over a wire or under another block.

---When it goes over a wire, often times clicking on the block will select the wire underneath it, and move or disconnect it. The block on top should always be selected, not the obscured part of the wire.

---When it goes under existing blocks, it's just hard to find.

screen capture of block over wire, but wire still gets selected.

[image]

I'm experiencing this in version 3.7.10.1.

Thanks,

NeuroForLunch, 2021-06-24:
I have attempted to replicate this and been unable to. Perhaps it has been fixed since Nov 2017?

marcusmueller, 2023-12-21:
Specific to 3.7 anyways,  since the whole connection logic has changed since then.

closed 2023-12-21 by marcusmueller