gnuradio/gnuradio #1516: Choppy sound from Audio Sink

Vort, 2017-11-08:
Hello.

I want to hear simple pulsing waves with help of GNU Radio Companion.
But pressing Execute button produces distorted sound in my headphones.
However, produced .wav file sounds good.

Here is my block diagram:
[image]

Related:
https://stackoverflow.com/questions/46797595/using-gnuradio-audio-sink-in-windows-enviornment

Ver: [gnuradio_3.7.11_win64_avx2.msi](http://www.gcndevelopment.com/gnuradio/downloads/installers/v1.3.0/gnuradio_3.7.11_win64_avx2.msi) ([website](http://www.gcndevelopment.com/gnuradio/downloads.htm))
OS: Windows 7 SP1 x64

Good sound, bad sound and .grc file are attached:
[audio_bad.zip](https://github.com/gnuradio/gnuradio/files/1452809/audio_bad.zip)

BrashEndeavours, 2017-11-09:
Could the combined amplitudes be clipping? Try turning them down.

Vort, 2017-11-09:
No, it does not helps.

Olivier-ADLER, 2017-11-10:
I had the same problem. Verified on two Intel PCs running Windows 7 fully updated. One with a dual core processor, the other with a six core intel I7 processor. The clicks did not come from the hardware audio, neither from the driver, because those two PCs internal audio hardware was working flawlessly with other softwares.

So after investigating, i think that something is broken or not correctly working inside GNURadio (3.7.11 win64) for synchronization between the audio card driver and the internal processing blocks.

In my case, using an audio sink alone block, GNURadio send to much samples to the audio driver, causing clicks and a wrong pitch. The difference in frequency is about 5%. Tested at 32 Khz and 48 khz the problem is the same.

If you disable the blocking for the audio sink block, you should see many overruns. In my case this high amount of overruns messages are causing a crash when blocking is disabled.

The solution to solve this problem is to add an audio source block, and send its output to the audio sink block, using for example a math "multiply constant" block set to "0" and a "add" block, so that you can keep the signal of your design entering in the second input of the "add" block without modifying it.

Perhaps you'll need to connect a microphone to the audio card Jack 3.5mm input, if not the audio source block could not work. In my case i did connect a stereo 3.5mm jack in the line input, to activate audio input and get the audio source block happy. On most modern computer if you don't do that audio input will not be available and will cause the audio source block to not work and report an error.

No more clicks, no more frequency offset, perfect sound without any distortion.

Hopefully this will work for you and will help the developers to find the root cause. Again, i think that it is simply a synchronization problem, perhaps because GNURadio does not talk to Windows to get a correct timer information (windows multimedia timer is known to not be precise) and / or does not talk correctly with the audio driver API.

I'm working since many years with audio applications under Windows, all working flawlessly. So the problem is certainly correctable but will need perhaps a conditional compilation for the Windows platform.

The method i did use :

[image]

Olivier-ADLER, 2017-11-10:
I should add that i my case, if i did record to a wav file, the audio was perfect. This is the reason why i did suspect a synchronization problem, and did solve it forcing GNURadio to run at the right audio sample rate adding an audio source block connected to the audio sink block.

dbnut, 2018-04-03:
Olivier, I'm a complete beginner with Gnu Radio Companion and was starting to think about retirement. After 4 days going crazy I found your solution. Brilliant. Explains everything, thank you so much.

marcusmueller, 2018-04-04:
Hey guys,

I'm very happy that this worked out for you! Keep up the good work!

So, Olivier's report worries me. I think there is something wrong with what Olivier does (or with what GNU Radio does on behalf of Olivier) if the /pitch/ changes.

 Namely, this indicates that the sound card is actually running at a different sample rate than assumed. 5% seems like an odd offset, but my guess would be that the soundcard driver, for whatever reason, in fact runs at 44.1 kHz when configured to 48 kHz. 

From my (hopefully fairly deep) understanding of how GNU Radio works, the "add" trick shouldn't really help - GNU Radio blocks are inherently slowed down to the rate the sink consumes, in the end. So, this indicates maybe a side effect of having an audio source?

Now, admittedly, I'm on vacation, and even if I weren't, I'd have no "real" windows to test this on myself (VM adds another unknown in the audio chain which is _very_ prone to problems).

Olivier-ADLER, 2018-04-04:
Hum,

The difference between 44.1 and 48 Khz is not 5% but around 10%.

More, i did test at 32 KHz and the problem was similar. I did test on two different PC with different (very common) integrated audio hardware from well known manufacturers.

The pitch change, i think, is not induced by a driver running at the wrong pitch, but by data sent at a wrong (higher) speed, causing choppy and very highly distorted sound at a wrong pitch.

The overruns messages we can see show that the sink consume speed and data speed are not equal (very different).

So according to me, GNU Radio is not slowed down by the sink block alone, seems like it does not manage correctly the data speed sent to the sink part of the audio driver for a reason that only GNU Radio developers will be able to find, probably.

Adding a source block seems to regulate audio data speed to the correct value, from source to sink. This seems logical to me, and this is because of this evidence that i did find this trick, as a former sound engineer used to work with digital audio hardware, where a digital source audio signal is often used to clock the sampling frequency of a full system, from input, to computing and output.

Another solution is to use a clock signal distributed to all audio equipments, but this is another story. This solution is used for large digital installations, to avoid propagation of clock jitter.

gnieboer, 2019-02-13:
The root cause has been found and the update code is in the branches awaiting a release.

The root cause was some blocks were emitting multiple chunks of data at a time and the audio sink assumed (incorrectly) that there was one chunk at a time.

UgoPoddine, 2019-05-23:
On Gnuradio 3.7.13.4 on Windows 7 PRO up-to-date the issue seems to be still present and the workaround above described solves the issue.

Olivier-ADLER, 2019-05-23:
Is the updated code included in the 3.7.13.4 Windows release ?

If it is included something is still broken.

A quick fix solution would be to include my previously described workaround in the code himself, or at least add a note in the audio sink module describing the workaround.

Will avoid a lot of troubles for users.

gnieboer, 2019-05-24:
It is not.  Expect an updated release after next weekend.

Faisalkhan087, 2019-11-03:
Hi, 
I have a problem in gnu radio that whenever i record and save a file it shows very large memory. for example i recorded 30 seconds song on FM radio on gnu radio using RTL-SDR dongle and it gives the memory file of 145.3 MB. How can i fix this problem ? how this memory storage can be reduced ?

Vort, 2019-11-03:
@Faisalkhan087 yes, I have noticed the same problem. It looks like memory leak exists somewhere. But that needs separate bugreport.

BrashEndeavours, 2019-11-03:
> @Faisalkhan087 yes, I have noticed the same problem. It looks like memory leak exists somewhere. But that needs separate bugreport.

@Faisalkhan087, are you talking about the size of the saved file? 
That's not a memory leak, IMO.

Faisalkhan087, 2019-11-03:
Yes I was talking about the file..the file that is made when we start the programme on gnu radio..that file is of large memory...it always comes in MB's as it is recorded in just few seconds..it should be in kB's because it's an mp3 file.. how can this problem be fixed ???Sent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/gnuradio/gnuradio/issues/1516?email_source=notifications\u0026email_token=ANU5GUA3OQDQ53KF4GP6QPTQR3QJRA5CNFSM4ECXU722YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEC5V5JQ#issuecomment-549150374",
"url": "https://github.com/gnuradio/gnuradio/issues/1516?email_source=notifications\u0026email_token=ANU5GUA3OQDQ53KF4GP6QPTQR3QJRA5CNFSM4ECXU722YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEC5V5JQ#issuecomment-549150374",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Vort, 2019-11-03:
@Faisalkhan087 if you are using Wav File Sink, then it is .wav, and not .mp3.
Size can be estimated by calculating sample rate * bit depth * time * channel count / 8.
For example, if you recorded 30 seconds of 200 kHz I/Q signal with bit depth = 16, then file size = 200000 * 16 * 30 * 2 / 8 = 24000000 = 24 Megabytes.

Faisalkhan087, 2019-11-04:
yeah that's right to calculate the size of file ?
actually i want to ask that how this file size can be minimized because if
i am recording it in minutes it should be in KB's.
further if i want to record the audio for whole day then it should be
maximum of 80 to 90 MB's.
further if i want to plot the graphs of that recorded audio data how it
should be done ???

Vort, 2019-11-04:
Bugtracker is not a place for such questions.

Vort, 2019-11-04:
I have tested program from 1st post with `gnuradio_3.7.13.5_win64_avx2.msi` and it works fine now.
Thanks for the fixing.

closed 2019-11-04 by Vort

fengshao1370, 2020-02-12:
> Hi,
> I have a problem in gnu radio that whenever i record and save a file it shows very large memory. for example i recorded 30 seconds song on FM radio on gnu radio using RTL-SDR dongle and it gives the memory file of 145.3 MB. How can i fix this problem ? how this memory storage can be reduced ?

hello，I am using GNURadio on Linux, but I meet the same question ,I installed a new version of gnuradio_3.7.13.5, the problem is still exsist. I just use a single sine signal of 500Hz. Doesn't it fix the problem? When I excute the program, it will output some auauau …letters. The sample rate of the audio sink must be  44100Hz?  Does it mean there must be some problems if it outputs aUaUaU?
thanks 

>

gnieboer, 2020-02-12:
I recommend you refer your question to the GNURadio mailing list.  From
what you are describing, this is likely expected behavior, but if you post
your flowgraph to the mailing list, they will be able to assist you.  If
then it turns out to be a bug, then you can open a new bug here in the
repository.

glangsto, 2021-04-13:
Hi.  Thanks for your hint.  I confirm that this fix of adding the audio source also worked for me, removing the annoying glitches in the Audio out.

iQyan, 2021-08-24:
I found the solution. You must change the Device Name field in Audio Sink block to “plughw:0,0”, as described in Documentation tab.