gnuradio/gnuradio #1518: Filter Design Tool does not open on Windows 7 x64.

Olivier-ADLER, 2017-11-10:
This tool does not open on Windows x64. Seems to be related to the graphic application only, because filters are working.

Ka-zam, 2017-11-13:
It doesn't open on Ubuntu 17.04 x64 either

katutxakurra, 2018-07-04:
In Ubuntu 16.04 I installed GNURadio using pybombs. The Filter Design Tool didn't open either. Running the script from the command line, I got the error
user@pc:~/src/gnuradio/default$ python ./lib/python2.7/dist-packages/gnuradio/filter/filter_design.py
Please install PyQwt5 to run this script (http://pyqwt.sourceforge.net/)
user@pc: sudo apt install python-qwt5-qt4
Solved the problem and the tool starts from command line and GNU Radio Companion (version 3.7.12git-361-g713629cc)
I guess that the pybomb recipe needs to add python-qwt5-qt4 as a dependency?

devnulling, 2020-01-19:
I'm going to close this issue as the filter design tool has been overhauled with the 3.8 release. I'm running into some new issues with it, so I'll make a new issue for those specifically.