gnuradio/gnuradio #1520: Variables not being recognized

Brett99, 2017-11-15:
I have a simple flowchart here, and the `samp_rate` variable is not being recognized by the blocks. 
[image]
Am I doing something wrong, or is this a GRC bug?

Brett99, 2017-11-15:
I can get the blocks to recognize `samp_rate` by deleting it and manually adding it again.

bastibl, 2017-11-15:
The gray blocks are disabled. Enable them with the corresponding option in the menu bar or by pressing 'e'.

Brett99, 2017-11-15:
Wow, thanks. I've done this twice now. Perhaps there should be some kind of warning printed so new users like me have a more helpful error message when a block is disabled.

closed 2017-11-15 by Brett99