gnuradio/gnuradio #1521: NameError: name 'gr' is not defined

Brett99, 2017-11-15:
I've installed gnuradio-companion on Arch, and I am having issues running nearly everything. I have connected a signal source to an audio sink, and when I try to compile it, I get this: 

```
Generating: '/home/brett/gnuradio/top_block.py'

Executing: /usr/sbin/python2 -u /home/brett/gnuradio/top_block.py

Traceback (most recent call last):
  File "/home/brett/gnuradio/top_block.py", line 24, in <module>
    class top_block(gr.top_block, Qt.QWidget):
NameError: name 'gr' is not defined

>>> Done
```

Brett99, 2017-11-15:
I created a new grc file with an identical flow chart, and it works for some reason. I now have two different files with identical flowcharts. One of them produces a working python file, and one does not. The only difference in the two files is that the working one includes these lines: 

```
from gnuradio import eng_notation
from gnuradio import gr
from gnuradio.eng_option import eng_option
from gnuradio.filter import firdes
from optparse import OptionParser
```
What could have happened to the first grc file to break it like this?

bastibl, 2017-11-15:
I assume you disabled the top block. You can enable it with the corresponding option in the menu bar or by pressing 'e'.

Brett99, 2017-11-15:
Thank you bastibl, I must've pressed d by accident. I ended up figuring it out by looking at the diff between the two files.

closed 2017-11-15 by Brett99