gnuradio/gnuradio #1523: eng_notation module / Parameter block: floating point parameters experience significant roundoff

mrjacobagilbert, 2017-11-15:
When using a Parameter block to pass a floating point value, the default value is formatted using `eng_notation.num_to_str()` which appears to limit the number of significant figures to 6. This can be problematic when (for example) using this for passing a non-default frequency. For frequencies in the GHz this can cause several kHz of error during roundoff.

The simple fix would be to remove the formatter for default float parameters (not sure why it's there) but the better fix is probably to address the num_to_str() function in the [eng_notation module](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/python/gnuradio/eng_notation.py). We want to do both.

marcusmueller, 2018-02-03:
So, as I just closed that PR

The problem is a very very valid one. I think we should be talking about how to flexibly do this! So, I could imagine having a precision parameter, or actually: Python is full of magic, and we might simply invent an operator that does what you need; imagine `1235657777.6776777777 % eng(precision)` to give you a string with given precision?

marcusmueller, 2018-02-03:
I'm *assigning* this issue to you, which means I entrust you with finding a solution that fulfills the requirements of the project (sadly, your first one didn't do that well enough :( ). So, this is actually more responsibility with the same workload for you, hope you don't mind!

mrjacobagilbert, 2018-02-03:
Marcus, I can appreciate your reasons for closing this. 

I am proficient with making software tweaks, but in full disclosure I am an RF guy (and scared of most of python's magic...) and would likely not do well at attempting to implement your specific suggestion.

My other idea was to add an optional argument to the num_to_str(n) function like this:

```def num_to_str (n, precision=6)```

Which should preserve the original functionality, and in places where it is not intended to be a user-friendly string could be extended to additional digits. Is that something you would see as a more appropriate solution?

mrjacobagilbert, 2018-02-03:
Also, i think its worth pointing out that I am attempting to solve a symptom here, the real issue is probably that a 'human-readable' formatting function is being applied on data that is machine-processed downstream. But thats a larger discussion.

noc0lour, 2018-02-03:
So one thing we can tackle on the way is converting this to `String.format()` syntax which should simplify using variables inside of curly braces in strings (and keeps code forward compatible `PEP 3101`).

Maybe there is some use in creating a custom formatter, something along the lines of https://tobywf.com/2015/12/custom-formatters/

mrjacobagilbert, 2018-02-03:
Yes, ```String.format()``` is along the lines I was thinking. If that is acceptable I'll get a PR together.

marcusmueller, 2018-02-11:
Closing this issue as you've sorted it out in #1595. :tada:

closed 2018-02-11 by marcusmueller