gnuradio/gnuradio #1526: AttributeError: 'file_source_sptr' object has no attribute 'set_begin_tag'

shiyan430, 2017-11-17:
Hi! I have a grc that runs well with a USRP on laptop (Gnuradio version 3.7.12git-295-ga0adcd33) , but reported the following error on a Respray Pi 3 (Embedded Linux, Gnuradio version 3.7.10.2). I have installed all required modules. How should I fix it? I am using the USRP B200mini.

linux; GNU C++ version 6.3.0; Boost_106300; UHD_003.010.002.000-release

-- Loading firmware image: /usr/share/uhd/images/usrp_b200_fw.hex...
-- Detected Device: B200mini
-- Loading FPGA image: /usr/share/uhd/images/usrp_b200mini_fpga.bin... done
-- Operating over USB 2.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Asking for clock rate 51.200000 MHz... 
-- Actually got clock rate 51.200000 MHz.
-- Performing timer loopback test... pass
-- Asking for clock rate 51.200000 MHz... OK
Traceback (most recent call last):
  File "TX_GS.py", line 146, in <module>
    main()
  File "TX_GS.py", line 135, in main
    tb = top_block_cls()
  File "TX_GS.py", line 68, in __init__
    self.blocks_file_source_0.set_begin_tag(pmt.PMT_NIL)
AttributeError: 'file_source_sptr' object has no attribute 'set_begin_tag'

drmpeg, 2017-11-18:
In your .grc file, delete these lines:

    <param>
      <key>begin_tag</key>
      <value>pmt.PMT_NIL</value>
    </param>

This is a new feature in 3.7.12 that's not backward compatible to 3.7.10.2.

marcusmueller, 2023-12-21:
3.7 era –> closing

closed 2023-12-21 by marcusmueller