gnuradio/gnuradio #1531: Recommended install via PyBombs fails on macOS High Sierra

nsbruce, 2017-11-22:
# System versions
System version: macOS 10.13.1
Package manager: Homebrew (due to an OOT module needing PythonNet, macports can't be used)
Python version: Python 2.7.14

# How to recreate
After creating an installation prefix (step 4 of the quick-start):
```
$ pybombs install gnuradio
```

# Output
```
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Password:
Collecting PyGTK
  Using cached pygtk-2.24.0.tar.bz2
    Complete output from command python setup.py egg_info:
    ********************************************************************
    * Building PyGTK using distutils is only supported on windows. *
    * To build PyGTK in a supported way, read the INSTALL file.    *
    ********************************************************************

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-build-OyRkiY/PyGTK/
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
PyBOMBS.install_manager - ERROR - Package has no install method: libevent
```

devnulling, 2020-01-21:
Closing this issue as it related to pybombs. and have created a new issue over on that repo.

closed 2020-01-21 by devnulling

michaelld, 2020-01-21:
@nsbruce Is pythonnet this: https://pythonnet.github.io ...? If so, I'll see if I can add it to MacPorts.

nsbruce, 2020-01-21:
Hi @michaelld that is the one, yes!