gnuradio/gnuradio #1532: How to make QT GUI Entry module support Chinese？

bighold, 2017-11-23:
I make a change on qtgui_entry_xml. As bellow：

  `24 lambda: self.set_$(id)($(type.conv)(str(self._$(id)_line_edit.text().toLocal8Bit()))))`
[image]

  Now I type in Chinese in the entry box of the Entry module. But after typing Enter, the Chinese in the input box will become garbled. But one of my display modules correctly displays the Chinese sent by the Entry module. Therefore, the question is how to ensure that the input box of the Entry module does not occur garbled.
[image]
[image]

jasonuher, 2021-02-23:
I think the move to python3 may have fixed this.  I believe the correct method for solving this issue was to ensure the text string was utf8 encoded prior to the call to 'str' (which happens by default in python3).  Here's my screenshot of the same scenario (well.. 中国 instead of 中人 in the original screenshot but...).  

A few thoughts:
- I'm not sure what the original motivation was for adding the "toLocal8bit" but I don't think that's necessary here
- Maybe this is a maint-3.8 issue only now?
- Maybe it depends on which language packs are installed though, too...?

(The attached flow graph is just a "qt gui entry" and a "qt gui label" linked to the same variable)

[image]

marcusmueller, 2023-12-21:
It's supported! I was really upset at some point that I couldn't use my last name in a field…

closed 2023-12-21 by marcusmueller