gnuradio/gnuradio #1534: WAV file sink only supports 8 and 16 bit data

Ka-zam, 2017-12-03:
This feels a bit unsatisfactory. I have an implementation ready that supports 8,16, 24, and 32 bit integer format as well as 32-bit floating point. Floats may be clamped to [-1,1].

I also noticed there were some subtle bugs in the float->int conversion. These are fixed but at the expense of a move from int to double conversion constants. In my opinion it's better to know things are right rather than to save a few cycles.

Should I submit this pull request?

mbr0wn, 2017-12-21:
This sounds like two PRs? WAV source, and float->int conversion? In any case, yes please, we'd really appreciate that!

Ka-zam, 2017-12-22:
Great, I'll submit a PR that implements theses changes for WAV source and sink. Just have to clean up and make sure I didn't leave any of my comments in there.

devnulling, 2020-01-19:
@Ka-zam ping on this , it'd be great to see a PR for these changes.

Ka-zam, 2020-01-28:
Unfortunately this code was lost in a disk crash. Yeah, I know, not a great idea to work on a local copy of the source tree. I still remember what I did, let me see if I can find time to implement again.

mbr0wn, 2021-05-14:
@drmpeg the libsndfile update should fix this right? I'm going to close it, I think this is no longer an issue.

closed 2021-05-14 by mbr0wn

drmpeg, 2021-05-14:
Yes, all taken care of and more.

NeuroForLunch, 2021-05-25:
@drmpeg which commits can I cherry pick to enable reading of more WAV files?

drmpeg, 2021-05-25:
The initial commit is https://github.com/gnuradio/gnuradio/commit/18f64ba685cfc2533e2d7726aa43c524b007d97b

And then there's a performance commit. https://github.com/gnuradio/gnuradio/commit/8f3d61984ecd9ecc97789a174a54c7f0505bddb4

The target for those commits was 3.9 with pybind11, so it's not a direct backport to 3.8, but it could be done. Perhaps upgrading to 3.9 would be easier?

NeuroForLunch, 2021-05-25:
Some OOT modules don't work with 3.9 because they still use swig. Hopefully it's not too troublesome to make this work.

NeuroForLunch, 2021-05-25:
Any help would be appreciated.

dkozel, 2021-05-25:
@NeuroForLunch It's probably easier to port the OOT modules to 3.9 rather than backporting these commits to 3.8. We're not going to be back-porting the new wav files to 3.8.

https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide

NeuroForLunch, 2021-05-26:
@dkozel it seems I spoke too soon. Most of the OOT modules are still at 3.8 including gnuradio organization's gr-bokehgui (https://github.com/gnuradio/gr-bokehgui/issues/33) and gr-inspector because most people are struggling to port from swig to pybind11.