gnuradio/gnuradio #1536: Q: why timer used for fft decimation

dernasherbrezon, 2017-12-11:
Hi!

I found this in the source code:
https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/lib/waterfall_sink_f_impl.cc#L541

Is it a correct way for throttling/decimating fft? 

I would expect skip samples based on sampleRate and update_time ( https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/lib/waterfall_sink_f_impl.cc#L177 )

noc0lour, 2018-01-14:
From what I see here the timer is not used to skip calculated values but to limit the update rate to the GUI. It will still display all calculated FFTs as long as the display_time variable is small enough and your PC can keep up with your hardware.

dernasherbrezon, 2018-01-15:
Right, but in-between there is a call to fft. So I would expect d_qApplication->postEvent call based on current timestamp, but fft update based on sample rate.

marcusmueller, 2018-01-27:
Hey, I'd hate to see this discussion go to waste, but:

This is a *bug tracker*, and you're discussing a *question*. This would be exquisitely well-suited for our  [mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio)!

mbr0wn, 2018-02-04:
Since the FFT sink is a sink, it has no obligation to process all samples. The current approach makes sense; it is a different approach from tools like fosphor though. Maybe this is a documentation issue, but the the point is that we set the update rate in order to throttle CPU utilization; any samples that aren't strictly necessary don't need to be processed can be dropped.

The ambiguity is in "strictly necessary". Since we have averaging, you could argue that we don't want to drop samples even between updates. At that point we circle back to the fact that the FFT sink is not designed for plotting all samples, it's more of an inspection tool.

Good question, though.

closed 2018-02-04 by mbr0wn

dernasherbrezon, 2018-02-05:
Thank you Martin! 

PS: I will raise question next time on stackoverflow.

marcusmueller, 2018-02-08:
@daniestevez this is a very GNU Radio-implementation specific question, and stackoverflow isn't the optimal place. In spite of the danger of repeating myself: the mailing list is your friend :)

Generally, we're not opposed to discussing GNU Radio-related things on SO, but really, if it's about "why was this thing done like that internally", better talk to the developer community on their platform, which *is* the mailing list.