gnuradio/gnuradio #1538: gnuradio version  3.7.11.1 Installation

Alhamzi, 2017-12-15:
Hallo
 I have a basic question. I am a beginner with GNURadio and I want to work on version 3.7.11.1 but when I install it through pybombs I get the latest version 3.7.12. How can I install a specific version like 3.7.11.1
thanks in advance

BrashEndeavours, 2017-12-17:
Edit the PyBombs config in ~/.pybombs/config.yml and add specific package flags for gnuradio to point to a specific commit to build from.

packages:
    gnuradio:
        gitbranch: Whateverbranch
        gitrev: 012345abc

mbr0wn, 2018-02-04:
@BrashEndeavours thanks for answering this! @Alhamzi the issue tracker is for tracking bugs and issues with the GNU Radio code base, so please direct such questions to the mailing list.

closed 2018-02-04 by mbr0wn