gnuradio/gnuradio #1539: error: make test failed

Alhamzi, 2017-12-16:
hallo
while building source file of GNU radio following error occurred during "make test"
98% tests passed, 5 tests failed out of 211

Total Test time (real) = 177.80 sec

The following tests FAILED:
	 85 - qa_polar_decoder_sc_systematic (Failed)
	 87 - qa_polar_decoder_sc_list (Failed)
	 89 - qa_polar_encoder_systematic (Failed)
	 90 - qa_polar_encoder (Failed)
	 92 - qa_polar_decoder_sc (Failed)
Errors while running CTest
Makefile:61: recipe for target 'test' failed
make: *** [test] Error 8

will this error create any problem in future while using gnuradio, henceforth should i continue or should i look to rectify this error
Any suggestion will be appreciated,
Thanks

drmpeg, 2017-12-18:
Most likely, you're missing scipy.

sudo apt-get install python-scipy

justtennis, 2019-11-20:
I see similar error messages after running: make test
...

_99% tests passed, 5 tests failed out of 363

Total Test time (real) = 152.15 sec

The following tests FAILED:
	243 - qa_polar_decoder_sc (Failed)
	244 - qa_polar_decoder_sc_list (Failed)
	245 - qa_polar_decoder_sc_systematic (Failed)
	246 - qa_polar_encoder (Failed)
	247 - qa_polar_encoder_systematic (Failed)
Errors while running CTest
Makefile:85: recipe for target 'test' failed
make: *** [test] Error 8_

I ran the suggested install for python-scipy (sudo apt-get install python-scipy) but still get these errors.

I am running a new install of Ubuntu 18.04.  
Any help would be greatly appreciated.

michaelld, 2019-11-20:
What does `ctest -V -R qa_polar_decoder_sc_list` return? Hopefully whatever it shows will make the solution obvious ...

drmpeg, 2019-11-21:
If you're building GNU Radio 3.8 or the master branch, you'll need the Python3 version instead.

sudo apt-get install python3-scipy

justtennis, 2019-11-21:
short answer: that fixed it for me.  Thanks!

@michaelld good call.  your suggested command returned "No module named 'scipy'"
@drmpeg I installed Python3-scipy as you suggested:
$sudo apt-get install python3-scipy
I reran $make test
100% tests passed (from $make test)
Looks like that fixed it for me.  Thanks for the quick and helpful responses.

michaelld, 2019-11-21:
You're welcome! Glad you got it all sorted out. Closing.

closed 2019-11-21 by michaelld