gnuradio/gnuradio #1541: Memory Allocation Bug

silidan, 2017-12-19:
gnuradio fails to allocate memory of size 1048576 KB or larger.
[memory_allocation_bug.grc.log](https://github.com/gnuradio/gnuradio/files/1572795/memory_allocation_bug.grc.log)

[image]
[memory_allocation_bug.log](https://github.com/gnuradio/gnuradio/files/1572783/memory_allocation_bug.log)

thasti, 2017-12-19:
To add some information to this:

- The flowgraph tries to create a vector with 1024*1024*64 complex values, which will cause a gr::buffer allocation for 2^30 elements. 
- gr::buffer implements this as a gr::vmcircbuf, with the size argument passed as an int (which is still okay so far). 
- gr::vmcircbuf_sysv_shm will try to allocate a shared memory buffer (shmget) with int(2 * size + 2 * pagesize) as an argument. This becomes greater than 2^31, overflows to negative number, which is in turn converted to size_t by being passed as an argument to shmget
- the cast to size_t causes the argument to become a huge positve number (close to 2^64 on 64 bit systems) and therefore being larger than either SHMMAX or the available system memory.

One can see multiple problems:
- gr::buffer should restrict the size of allocated buffers (maybe)
- gr::vmcircbuf checks for size to be negative, but then tries to shmget 2*size+2*pagesize, which might cause an overflow itself.
- in general, maybe size_t types should be used here and there.

marcusmueller, 2023-12-21:
vlosed per #4460

willcode, 2023-12-21:
The change to size_t should make it work up to 2GB, if the system can provide 2x that much consecutive virtual memory. Things will always fail for a high enough value.

closed 2023-12-21 by willcode