gnuradio/gnuradio #1544: Error on gr-trellis during make

c0debreaker, 2018-01-10:
```
[ 81%] Building CXX object gr-trellis/lib/CMakeFiles/gnuradio-trellis.dir/core_algorithms.cc.o
[ 81%] Building CXX object gr-trellis/lib/CMakeFiles/gnuradio-trellis.dir/fsm.cc.o
[ 81%] Building CXX object gr-trellis/lib/CMakeFiles/gnuradio-trellis.dir/interleaver.cc.o
[ 81%] Building CXX object gr-trellis/lib/CMakeFiles/gnuradio-trellis.dir/quicksort_index.cc.o
[ 81%] Building CXX object gr-trellis/lib/CMakeFiles/gnuradio-trellis.dir/constellation_metrics_cf_impl.cc.o
[ 81%] Building CXX object gr-trellis/lib/CMakeFiles/gnuradio-trellis.dir/permutation_impl.cc.o
[ 81%] Building CXX object gr-trellis/lib/CMakeFiles/gnuradio-trellis.dir/siso_f_impl.cc.o
[ 82%] Building CXX object gr-trellis/lib/CMakeFiles/gnuradio-trellis.dir/siso_combined_f_impl.cc.o
[ 82%] Linking CXX shared library libgnuradio-trellis.dylib
[ 82%] Built target gnuradio-trellis
Scanning dependencies of target gr_trellis_xml
[ 82%] Generating test_tcm.py.xml
[ 82%] Generating test_viterbi_equalization1.py.xml
[ 82%] Built target gr_trellis_xml
Scanning dependencies of target gr_trellis_html
[ 82%] Generating gr-trellis.html
xmlto: /Users/neil/repos/experiment/gnuradio-3.7.11/gr-trellis/doc/gr-trellis.xml does not validate (status 3)
xmlto: Fix document syntax or use --skip-validation option
/Users/neil/repos/experiment/gnuradio-3.7.11/gr-trellis/doc/gr-trellis.xml:6: warning: failed to load external entity "/Users/neil/repos/experiment/gnuradio-3.7.11/gr-trellis/doc/docbookx.dtd"
]>
  ^
warning: failed to load external entity "/Users/neil/repos/experiment/gnuradio-3.7.11/gr-trellis/doc/docbookx.dtd"
validity error : Could not load the external subset "docbookx.dtd"
Document /Users/neil/repos/experiment/gnuradio-3.7.11/gr-trellis/doc/gr-trellis.xml does not validate
make[2]: *** [gr-trellis/doc/gr-trellis.html] Error 13
make[1]: *** [gr-trellis/doc/CMakeFiles/gr_trellis_html.dir/all] Error 2
make: *** [all] Error 2
```

c0debreaker, 2018-01-10:
I fixed it.

closed 2018-01-10 by c0debreaker

marcusmueller, 2018-02-11:
Hi @c0debreaker, what does "I fixed it" mean? Does it mean you broke something on your own system and reverted that, or does it mean you found a fix for a problem with GNU Radio itself that you could share?

c0debreaker, 2018-02-15:
I should be detailed next time. I'm really sorry I just wrote `I fixed it`. Now I'm having a hard time trying to recall what I did. Most likely I just installed a missing dependency.

c0debreaker, 2018-02-15:
This might help since I documented it. I think this is the gist that's related to this ticket
https://gist.github.com/c0debreaker/252e5e179bc5e7ca76c4041454703314

marcusmueller, 2018-02-16:
Thanks! This is an issue I myself had an forgot to raise here. Will make this an official thing. Thanks!

boushley, 2022-01-16:
I'm running into this same problem 4 years later :)

The gist suggested doesn't seem to have anything applicable to this error that I can understand.

I've pushed pass the error by commenting out [the build of this specific help/html file](https://github.com/gnuradio/gnuradio/blob/3d4d9a4245b9e126d7119a7c81905c0d5423233b/gr-trellis/docs/CMakeLists.txt#L15-L28). Seeing that this comes from running `xmlto` is it possible there is an issue with the version of `xmlto` that I'm running? I'm running `0.0.28` which appears to be the latest version from 2015.