gnuradio/gnuradio #1545: Failed more than 94% on make test

c0debreaker, 2018-01-10:
The `make` was fine but the make test failed many tests.
Here is the output

```
 ❯ make test
Running tests...
Test project /Users/rcamara/repos/experiment/gnuradio-3.7.11/build
        Start   1: qa_test_all
  1/212 Test   #1: qa_test_all ..........................   Passed    0.71 sec
        Start   2: gr-runtime-test
  2/212 Test   #2: gr-runtime-test ......................   Passed    4.27 sec
        Start   3: gr-pmt-test
  3/212 Test   #3: gr-pmt-test ..........................   Passed    1.53 sec
        Start   4: qa_feval
  4/212 Test   #4: qa_feval .............................***Failed    0.05 sec
        Start   5: qa_flowgraph
  5/212 Test   #5: qa_flowgraph .........................***Failed    0.32 sec
        Start   6: qa_hier_block2
  6/212 Test   #6: qa_hier_block2 .......................***Failed    0.22 sec
        Start   7: qa_kludged_imports
  7/212 Test   #7: qa_kludged_imports ...................***Failed    0.04 sec
        Start   8: qa_random
  8/212 Test   #8: qa_random ............................***Failed    0.04 sec
        Start   9: qa_tag_utils
  9/212 Test   #9: qa_tag_utils .........................***Failed    0.04 sec
        Start  10: qa_pmt
 10/212 Test  #10: qa_pmt ...............................   Passed    0.16 sec
        Start  11: qa_pmt_to_python
 11/212 Test  #11: qa_pmt_to_python .....................   Passed    0.15 sec
        Start  12: test_gr_blocks
 12/212 Test  #12: test_gr_blocks .......................   Passed    1.12 sec
        Start  13: qa_add_mult_div_sub
 13/212 Test  #13: qa_add_mult_div_sub ..................***Failed    0.04 sec
        Start  14: qa_add_mult_v
 14/212 Test  #14: qa_add_mult_v ........................***Failed    0.05 sec
        Start  15: qa_affinity
 15/212 Test  #15: qa_affinity ..........................***Failed    0.04 sec
        Start  16: qa_argmax
 16/212 Test  #16: qa_argmax ............................***Failed    0.04 sec
        Start  17: qa_bin_statistics
 17/212 Test  #17: qa_bin_statistics ....................***Failed    0.04 sec
        Start  18: qa_block_behavior
 18/212 Test  #18: qa_block_behavior ....................***Failed    0.04 sec
        Start  19: qa_block_gateway
 19/212 Test  #19: qa_block_gateway .....................***Failed    0.16 sec
        Start  20: qa_boolean_operators
 20/212 Test  #20: qa_boolean_operators .................***Failed    0.04 sec
        Start  21: qa_burst_tagger
 21/212 Test  #21: qa_burst_tagger ......................***Failed    0.04 sec
        Start  22: qa_conjugate
 22/212 Test  #22: qa_conjugate .........................***Failed    0.04 sec
        Start  23: qa_copy
 23/212 Test  #23: qa_copy ..............................***Failed    0.04 sec
        Start  24: qa_delay
 24/212 Test  #24: qa_delay .............................***Failed    0.04 sec
        Start  25: qa_endian_swap
 25/212 Test  #25: qa_endian_swap .......................***Failed    0.04 sec
        Start  26: qa_file_metadata
 26/212 Test  #26: qa_file_metadata .....................***Failed    0.04 sec
        Start  27: qa_file_source_sink
 27/212 Test  #27: qa_file_source_sink ..................***Failed    0.04 sec
        Start  28: qa_head
 28/212 Test  #28: qa_head ..............................***Failed    0.04 sec
        Start  29: qa_hier_block2
 29/212 Test  #29: qa_hier_block2 .......................***Failed    0.04 sec
        Start  30: qa_hier_block2_message_connections
 30/212 Test  #30: qa_hier_block2_message_connections ...***Failed    0.05 sec
        Start  31: qa_integrate
 31/212 Test  #31: qa_integrate .........................***Failed    0.04 sec
        Start  32: qa_interleave
[... 544 lines cut ...]
	188 - qa_simple_framer (Failed)
	189 - qa_dtv (Failed)
	191 - qa_atsc (Failed)
	192 - qa_channel_model (Failed)
	193 - qa_fading_model (Failed)
	194 - qa_noaa (Failed)
	195 - qa_pager (Failed)
	196 - qa_trellis (Failed)
	198 - qa_video_sdl (Failed)
	199 - qa_alaw_vocoder (Failed)
	200 - qa_codec2_vocoder (Failed)
	201 - qa_cvsd_vocoder (Failed)
	202 - qa_g721_vocoder (Failed)
	203 - qa_g723_24_vocoder (Failed)
	204 - qa_g723_40_vocoder (Failed)
	205 - qa_gsm_full_rate (Failed)
	206 - qa_ulaw_vocoder (Failed)
	207 - qa_fcd (Failed)
	208 - qa_classify (Failed)
	209 - qa_zeromq_pub (Failed)
	210 - qa_zeromq_pubsub (Failed)
	211 - qa_zeromq_pushpull (Failed)
	212 - qa_zeromq_reqrep (Failed)
Errors while running CTest
```

c0debreaker, 2018-01-10:
I fixed it after installing scipy. :)

marcusmueller, 2018-02-11:
Changing title, reopening: this is actually an issue, and unless you submitted a pull request with a fix for that issue (namely, need for scipy not tracked for `make test`), it shouldn't be closed.

noc0lour, 2018-02-11:
Replacing scipy with something else already present would be a viable solution imho.

marcusmueller, 2018-02-11:
I'd argue that if someone builds GNU Radio, then having scipy as *build-time* dependency (just like CMake, SWIG, …) really isn't much of a burden. Packaging systems track these separately from run-time deps, anyway.

The question is whether it should be a run-time dependency, too, is a different one. None of our `gr_plot_**` tools work without, nor does `gr_filter_design`, and a lot of the gr-digital and -dtv and -filter examples depend on it, too. In many cases, I'd agree, all that's used is `fftpack`.... which does come with numpy.

jdemel, 2018-02-18:
SciPy use should actually be reduced to a minimum. Here in paticular, polar codes actually use specific SciPy function. 
I agree that making SciPy a build/run-time requirement does not hurt. Especially because almost everyone will need it and it is quite tightly bundled with NumPy anyways.

marcusmueller, 2018-02-18:
Just having had a night of rest about this: don't underestimate the effort that goes into packaging Python packages for embedded platforms. If we do make scipy a runtime req, we should really very clearly define the affected components -maybe even in finer granularity than "gr-utils".

jdemel, 2018-02-19:
I had this discussion in mind, when I added a comment to GREP0001.
The SciPy import should be in a try/except clause and the corresponding code should offer a fallback solution and a warning in case SciPy is not available.
I assume this'll fix these test case failures.

nickoe, 2020-03-30:
It looks like the fixes mentioned above are on all branches now, master, maint-3.7 and maint-3.8. I think we can close this relic.

closed 2020-03-30 by nickoe