gnuradio/gnuradio #1546: sdrplay with gnuradio

Mitchell1314, 2018-01-12:
Can sdrplay RSP2 work with gnuradio? Can you give me some help about how i can do it?

willcode, 2018-01-12:
Yes, I'm just about finished with better support for the SDRplay devices 
in gr-osmosdr. You can try out the current state by pulling

https://github.com/willcode/gr-osmosdr

and checking out the "sdrplay2" branch.

This code is almost ready to be merged. There is also support through Soapy.

willcode, 2018-01-12:
Further instructions to help test this out:

install the latest SDRplay SDK (v2.11 15NOV2017)

git clone https://github.com/willcode/gr-osmosdr
cd gr-osmosdr
git checkout sdrplay2
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/tmp/test -DENABLE_NONFREE=yes ..

(make sure SDRplay shows up in the enabled list)

mkdir /tmp/test
make install
LD_LIBRARY_PATH=/tmp/test/lib64:$LD_LIBRARY_PATH gqrx

marcusmueller, 2018-01-27:
Hi Mitchell,

as willcode very helpfully pointed out, this is an issue for the gr-osmosdr module.

Anyway, please understand that this is an *issue tracker*; you asked a question. This is not the right place to ask questions, please refer to out [mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio), or our IRC channel (`#gnuradio` on freenode), or our slack channel (`#gnuradio` on gnuradio.slack.com).

closed 2018-01-27 by noc0lour