gnuradio/gnuradio #1548: blocks: vector_sink reserve (min) data length in constructor

noc0lour, 2018-01-14:
In the `vector_sink` block data is allocated dynamically on every call. To improve performance a user could supply a minimum number of items he plans to stream into the vector sink, so memory is already allocated. Could improve speed during testing as well.

marcusmueller, 2018-01-27:
Idea: Make vector_sink have a "total number of items to store": Vector sink is in most cases used directly after a head block, and that's very a) annoying, b) a copying of perfectly fine samples and c) would be an excellent size to allocate first.

noc0lour, 2018-01-27:
So, basically have a `num_items` ?

lshariprasad, 2026-03-17:
Hi, I'm Hari Prasad L S, a GSoC 2026 applicant. I'd like to work on this issue and plan to submit a PR within 5 days.