gnuradio/gnuradio #1553: cmake: old self-maintained useSWIG?

noc0lour, 2018-01-20:
What's the justification for not using `useSWIG.cmake` from upstream cmake?
In a newly created OOT in encountered issues up to not being able to build with `ninja`. Removal of `useSwig.cmake` in `cmake/Modules` solved that.

noc0lour, 2018-01-20:
OK, apparently copying the current `UseSWIG.cmake` to the OOT works out as well.
Using the upstream `UseSWIG.cmake` won't work for the GNU Radio tree though. Maybe we can upstream fixes?

marcusmueller, 2018-02-11:
@noc0lour any chance of a diff of upstream vs ours?

marcusmueller, 2018-02-11:
Also, do we agree that if we can count on a working `useSWIG.cmake` being part of the GNU Radio development headers one needs to have installed, anyway, can't we then just drop the `useSWIG.cmake` from gr-newmod? I'd consider a cruft-free OOT template a feature.

noc0lour, 2018-02-11:
Installing our own `useSWIG.cmake` will break things for package manager, so please don't do that. We can indeed install our `Gr*.cmake` Modules in `/usr/share/cmake/Modules` and stop shipping (old) versions of those in `gr_modtool`.

marcusmueller, 2018-02-11:
sorry, was ambiguous: What I meant is that if we can rely on GNU Radio-supplied CMake infra to either use a working system-installed useSWIG or supply the same code (in a non-conflicting way), then we shouldn't put any code that's in a GNU Radio installation into an OOT.

noc0lour, 2018-02-11:
Yes, I always wondered why we are not supplying `Gr*` cmake Modules and instead duplicate it for every Module...

noc0lour, 2018-02-11:
Diff between our `next` and CMake 3.9.6 supplied `UseSwig.cmake`

https://gist.github.com/noc0lour/3900b7563969c985ea59071e7d3f1a37

marcusmueller, 2018-02-11:
> I always wondered why we are not supplying Gr* cmake Modules and instead duplicate it for every Module...

Me too. Probably the fear back then<sup>TM</sup> was that we might retroactively break OOTs when updating our CMake-Foo. My fear now is that we can't update an OOT's CMake Scripts, since they are bundled.

marcusmueller, 2018-02-11:
ugh. that diff isn't as singular as I hoped it'd be. Would you share my assessment that it's probably better to make GR work with the upstream useSWIG than to ship this code (even if in module of a different name)?

noc0lour, 2018-02-11:
Need to check for the full range of `cmake` we want to support in `next` and then fix it there.

noc0lour, 2018-02-11:
Originally (in 2014) the custom cmake was imported because: `cmake: use custom UseSWIG.cmake to fix parallel build issue`. So need to figure out if that's still the case or if this is fixed upstream for all required versions)

marcusmueller, 2018-02-11:
Even if it's not, then we've got some upstreaming to do :)

noc0lour, 2018-02-11:
But still have to ship ours until it's fixed in every `cmake` version we want to support (:

mormj, 2021-03-09:
Closing since `swig` is removed in 3.9, and doubtful we want to change the range of supported `cmake` versions for 3.8

closed 2021-03-09 by mormj