gnuradio/gnuradio #1555: Problem with PyBombs and gnuradio

emozo, 2018-01-23:
Hello, I want to install gnuradio through the PyBOMBs, I am executing the follow comand and I have this problem, I need any suggestion about it, Thank you

erislandy@erislandy-VirtualBox:~$ pip install PyBOMBS
....
....
Successfully built PyBOMBS future
Installing collected packages: setuptools, certifi, chardet, idna, urllib3, requests, six, future, ruamel.ordereddict, ruamel.yaml, PyBOMBS
Successfully installed PyBOMBS certifi chardet future idna requests ruamel.ordereddict ruamel.yaml setuptools-20.7.0

erislandy@erislandy-VirtualBox:~$ pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Cloning:     (100%) [=========================================================]

erislandy@erislandy-VirtualBox:~$ pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Cloning:     (100%) [=========================================================]

erislandy@erislandy-VirtualBox:~$ pybombs prefix init ~/prefix/default/
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.prefix - ERROR - Cannot write to prefix path `/home/erislandy/prefix/default'.

willcode, 2018-01-23:
There are some initialization steps. Try

  pybombs prefix init /home/erislandy/prefix

emozo, 2018-01-23:
Hello willcode thank you for your fast answer, I could install very well gnuradio, it was the solution. By I can install the master branch of gnuradio but I want to install the next branch of gnuradio, What can I do ?

willcode, 2018-01-23:
Please close this issue - this is a help request, not a bug. As for using next, take a look at the recipe files.

emozo, 2018-01-23:
Ok thank you

closed 2018-01-23 by emozo