gnuradio/gnuradio #1565: cmake: cmake_c_flags are set wrong on next for ubuntu 14.04

noc0lour, 2018-01-29:
Apparently this: 
```
# Configure C++ and C standards
IF(CMAKE_VERSION VERSION_LESS "3.1")
    IF(CMAKE_CXX_COMPILER_ID STREQUAL "GNU")
        SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
    ELSEIF(CMAKE_CXX_COMPILER_ID STREQUAL "Clang")
        SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
    ELSEIF(CMAKE_CXX_COMPILER_ID STREQUAL "MSVC")
        SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
    ELSE()
        MESSAGE(warning "C++ standard could not be set because compiler is not GNU, Clang or MSVC.")
    ENDIF()

    IF(CMAKE_C_COMPILER_ID STREQUAL "GNU")
        SET(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=gnu99")
    ELSEIF(CMAKE_C_COMPILER_ID STREQUAL "Clang")
        SET(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=gnu99")
    ELSEIF(CMAKE_C_COMPILER_ID STREQUAL "MSVC")
        SET(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c11")
    ELSE()
        MESSAGE(warning "C standard could not be set because compiler is not GNU, Clang or MSVC.")
    ENDIF()
ELSE()
    SET(CMAKE_C_STANDARD 99)
    SET(CMAKE_CXX_STANDARD 11)
ENDIF()
```
has no effect on `Ubuntu 14.04` and leads to: 
```
-- Compiler Flags: /usr/lib/ccache/cc:::-O3 -DNDEBUG  -std=c11 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/lib/ccache/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
```

No idea what is happening there.... Maybe the `VERSION_LESS` with "3.1" isn't working and it has to be "3.1.0". But that's clearly a problem for cmake 2.8 people to solve.

noc0lour, 2018-01-29:
Which is causing build failures because `M_SQRT2` is not defined in `c11` but in the non-standard compatible `gnu99/11` extension.

noc0lour, 2018-01-29:
I was dumb:
This line is the offender:
```
   SET(CMAKE_CXX_STANDARD 98)
+    SET(CMAKE_CXX_STANDARD 11)
 ELSEIF(CMAKE_CXX_COMPILER_ID STREQUAL "Clang")
-    SET(CMAKE_CXX_STANDARD 98)
+    SET(CMAKE_CXX_STANDARD 11)
 ELSEIF(CMAKE_CXX_COMPILER_ID STREQUAL "MSVC")
-    SET(CMAKE_CXX_STANDARD 98)
+    SET(CMAKE_CXX_STANDARD 11)
 ELSE()
     message(warning "C++ standard could not be set because compiler is not GNU, Clang or MSVC.")
 ENDIF()
 
 IF(CMAKE_C_COMPILER_ID STREQUAL "GNU")
-    SET(CMAKE_C_STANDARD 99)
+    SET(CMAKE_C_STANDARD 11)
 ELSEIF(CMAKE_C_COMPILER_ID STREQUAL "Clang")
-    SET(CMAKE_C_STANDARD 99)
+    SET(CMAKE_C_STANDARD 11)
 ELSEIF(CMAKE_C_COMPILER_ID STREQUAL "MSVC")
     SET(CMAKE_C_STANDARD 11)
 ELSE()
```

Since commit 4ba906d03a6c06f43cfbd25cbd230f1d1a1e6c12 it's broken.

noc0lour, 2018-01-29:
using `-std=gnu${CMAKE_C_STANDARD}` it should work.

mbr0wn, 2018-02-01:
Wouldn't it make more sense to not use a non-standard value, and define `M_SQRT` ourself? More standard-compliant.

noc0lour, 2018-02-01:
We can do that as well. I'm wondering why it works on newer cmake/gcc versions though. Are they defaulting to gnu11 if CMAKE_C_STANDARD is set to 11?

mbr0wn, 2018-02-04:
Yeah, they do (it even says so in the manual). Why they would do that is beyond me. We might have to always set the C flags, and not rely on `CMAKE_C_STANDARD`.

noc0lour, 2018-02-04:
Hm, ok sucks, then we definitely have to throw out `M_*` defines and replace them with our own. Nothing else is affected I think.

noc0lour, 2018-02-05:
So, trying to fix this by adding required `#define`s in `<gnuradio/math.h>` results in the C files in `viterbi/metrics.c` to fail compilation (since `gnuradio/math.h` is a c++ header). Either I define them on place (which sucks for future use in other places) or we could them transform them into valid `C++` and use the `<gnuradio/math.h>` header to do definitions.

marcusmueller, 2018-02-11:
Notice that we won't be supporting Ubuntu 14.04 on next. Should we be closing this issue?

closed 2018-02-16 by marcusmueller

noc0lour, 2018-02-16:
Then we still should take care off cmake < 3.1 and the `-std=gnu/c` issue.

marcusmueller, 2018-02-17:
Didn't we take care of that in the "another stab..." pr?

noc0lour, 2018-02-17:
Ah, sorry, yes.