gnuradio/gnuradio #1568: [Socket PDU]: Socket connections not destroyed when closed by client

juan0fran, 2018-01-30:
Hi, in gr-blocks at socket_pdu_impl, it seams that no-socket is destroyed in TCP-Server mode when the connection is closed by the client. It can be easily seen by creating the Socket PDU and connect-close the socket with an external application. After 1128 (in my case) of connect-close actions, the socket pdu throws boost::system::system_error(error); with error code = 24, that is:

- asio::error::no_descriptors 24

willcode, 2018-01-30:
At first glance, it looks like newly accepted connections are added to d_tcp_connections and there is no way for the reference to be released. I can take a look at it if the fix isn't obvious to someone else.

willcode, 2018-01-31:
socket_pdu_impl keeps a list of (reference to) tcp_connections so it can broadcast on the send side. But, tcp_connection handles the recv side, and also knows when the connection closes. There is no way currently for socket_pdu_impl to know the connection has closed, so it maintains the reference, which keeps the tcp_connection destructor from getting called. I think it will take an API change to fix this. Nothing else in-tree is using tcp_connection, so changing its constructor to take a "closing" callback would probably not have much impact. I spent a bit of time looking through boost::asio to find a way to set a "closing" callback at that level (that could be set up during accept(), but found nothing. Opinions?

mbr0wn, 2018-02-04:
@willcode does this get fixed by #1570?

willcode, 2018-02-04:
Yes - the pull was a response to this issue.

marcusmueller, 2018-02-11:
In that case, closing this issue as fixed by #1570.

closed 2018-02-11 by marcusmueller