gnuradio/gnuradio #1576: Problem with next branch of gnuradio

emozo, 2018-02-03:
Hello, thank you for yor helping, but i have the follow problem. I have installed very well the Pybombs program and I want to install the next branch of the gnuradio ( have installed the master branch of gnuradio very well). For that reason I have configured the ~/.pybombs/config.yml file for taking the next branch like :

         recipes:
                    gr-recipes: git+https://github.com/gnuradio/gr-recipes.git
                    gr-etcetera: git+https://github.com/gnuradio/gr-etcetera.git
        config:
                   default_prefix: /home/erislandy/prefix
        packages:
                   gnuradio:
                                gitbranch: next
                                forcebuild: 'True'
After that, I use: pypombs install gnuradio , for installing gnuradio but the instalation fall, it take me the follow error

     PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
    PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
    Configuration failed
    PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
I do not know if I have configured well the file ~/.pybombs/config.yml or have to do anything. I appreciate any advice about my problem, thank

mbr0wn, 2018-02-04:
The config file looks fine. Run PyBOMBS with `-vv` and see where it fails, and what it does.

Since this doesn't look like a bug with the actual GNU Radio codebase, I'm closing this, but please discuss it on the mailing list.

closed 2018-02-04 by mbr0wn

marcusmueller, 2018-02-11:
This is a general reminder:

The original question reads as if it is not a *bug report about GNU Radio itself*. Notice that this is a *bug tracker*. The [discuss-gnuradio mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio) is the proper place to discuss approaches to solving complex problems. 

I'm really happy about every bug being reported here, even if it not turns out to be a bug. I just get this feeling that people in the future will use a search engine, find this page and comment on this  discussion: Please don't. Use the mailing list for discussions! It's much more active.

If you find a *bug* in PyBOMBS, please *do* report it (adding full background, i.e. version of pybombs, OS/distro, full output, usually even set to verboseness), in the PyBOMBS issue tracker: 

https://github.com/gnuradio/pybombs/issues/

The same goes for the recipes in gr-recipes.

https://github.com/gnuradio/gr-recipes/issues/

Note that finding the right place to report is usually the first step necessary to solve an issue, anyway!