gnuradio/gnuradio #1577: Modtool add is broken

mrjacobagilbert, 2018-02-03:
Not sure what happened, but b0912d31f1a2cfea93a5588898cae0c652cd89d4 broke `gr_modtool add`:

```
jacob@ubuntu:/opt/gnuradio/src/gr-pdu_utils$ gr_modtool add
GNU Radio module name identified: pdu_utils
('sink', 'source', 'sync', 'decimator', 'interpolator', 'general', 'tagged_stream', 'hier', 'noblock')
Enter block type: sync
Traceback (most recent call last):
  File "/opt/gnuradio/bin/gr_modtool", line 46, in <module>
    main()
  File "/opt/gnuradio/bin/gr_modtool", line 37, in main
    modtool.setup(options, args)
  File "/opt/gnuradio/lib/python2.7/dist-packages/gnuradio/modtool/modtool_add.py", line 83, in setup
    print 'Must be one of ' + str(self._block_types)
TypeError: __exit__() takes exactly 1 argument (4 given)
```

noc0lour, 2018-02-03:
Looks like the signature for `__exit__` is wrong, it has to accept 3 arguments (+ self) to be called. https://docs.quantifiedcode.com/python-anti-patterns/correctness/exit_must_accept_three_arguments.html

marcusmueller, 2018-02-04:
Since that's my fault, I'll fix this

marcusmueller, 2018-02-04:
By the way, interestingly, this always works for me! So, while @noc0lour is absolutely right and I neglected to accept the three exception describing arguments, this should actually only be an issue if there's an exception raised in the `with` block.

So, I'll patch the `__exit__` method, but chances are, something else is still broken.

marcusmueller, 2018-02-05:
@mrjacobagilbert I've fixed the immediate issue (thanks, @noc0lour for pointing out my mistake!), but I don't believe that was the original cause for the problem. So, if you could try again with the fix I just merged into master, that'd be very appreciated.

closed 2018-02-05 by marcusmueller

mrjacobagilbert, 2018-02-05:
5473d529f69810bbb8f275779a188b47bdbd2002 solves the issue I observed.

mbr0wn, 2018-02-05:
Might be broken on next then too.

mrjacobagilbert, 2018-02-06:
Ok... gr_modtool rm is broken also...

```
jacob@ubuntu:/opt/gnuradio/src/gr-pdu_utils$ gr_modtool rm
GNU Radio module name identified: pdu_utils
Traceback (most recent call last):
  File "/opt/gnuradio/bin/gr_modtool", line 46, in <module>
    main()
  File "/opt/gnuradio/bin/gr_modtool", line 37, in main
    modtool.setup(options, args)
  File "/opt/gnuradio/lib/python2.7/dist-packages/gnuradio/modtool/modtool_rm.py", line 49, in setup
    with SequenceCompleter():
TypeError: __init__() takes exactly 2 arguments (1 given)
```

noc0lour, 2018-02-06:
That one needs a smarter complete function to take existing block names into account, or just no complete function.

mrjacobagilbert, 2018-02-06:
@marcusmueller - do you want me to open a separate issue for this?

marcusmueller, 2018-02-08:
@mrjacobagilbert yes, please.