gnuradio/gnuradio #1579: Hardware Impairment Block adds phase noise regardless of configuration

dkozel, 2018-02-04:
The Hardware Impairment block adds a lot of phase noise to the signal regardless of the Phase Noise value being set to 0.

Blue is the input signal, Red is the output of the HW impairments block.

[image]

dkozel, 2018-02-05:
Theory from last night, the setting of "0" on the phase noise is 0 dBc.

noc0lour, 2018-02-05:
Does it make sense to give the value in `dB` and not in linear scale?

mbr0wn, 2021-05-14:
I took a look and it is the noise power (in dB) of the analog noise source that is used to generate faux phase noise. Pretty convoluted. We can probably do better. But I think just improving the label for this will be sufficient.

marcusmueller, 2021-05-16:
Bug: Hardware impairment block brings undesired hardware impairment

Mental Category: Hyperrealism.

S,CNR.

closed 2021-05-17 by mormj