gnuradio/gnuradio #1583: Python block should be able to accept a variable number of input ports

giladbeeri, 2018-02-06:
C++ blocks can accept a variable input of ports when passing -1 as the maximum number of ports.
Python blocks don't seem to have this ability due to the way gateway_block() works.

willcode, 2018-02-07:
Since you can make up any signature you want when you create a Python block, e.g.,

g=gr.gateway.basic_block("test", [float,float],[float])

what is it that you're unable to do? If you create a class the derives from gateway.basic_block, it can set the number of inputs when the block is instantiated.

giladbeeri, 2018-02-07:
Exactly

willcode, 2018-02-07:
@giladbeeri are you saying that the existing gateway does what you need?

giladbeeri, 2018-02-07:
No, I agree with your analysis of gateway.py. The problem is that it accepts a list, and maps the list to numpy.dtype `sig_to_dtype_sig()`, and later `sig_to_gr_io_sigv()` uses `len(sig)` as to declare the maximum number of ports when calling `io_signaturev()`.

In C++ blocks, the API allows me passing -1 as the number of ports, which is used in ports like add and max (a variable number of input ports). In Python I have to pass a list in a definite length upon construction. I think some manipulation of `sig_to_dtype_sig()` and `sig_to_gr_io_sigv()` to allow `io_signaturev()` to be called with -1 is required. Also looks like `__in_indexes` will have to be updated when connecting additional blocks.

willcode, 2018-02-07:
@giladbeeri you closed the issue, so I was wondering if you were all set. I think you can reopen it if you want. My thought was that __init__ would accept either a list of types as it does now, or io_signature as arguments, to keep backward compatibility. The -1 would be in the io_signature. The trick is in figuring out whether the optimizations in gateway.py will be affected. Where io_signature is passed in, __in_indexes would probably not be used.

giladbeeri, 2018-02-07:
by mistake :)

Your solutions sounds like a good fit.

marcusmueller, 2018-02-11:
um, can you link to that solution? I'm a little confused here :)

willcode, 2018-02-11:
"solution" == "idea of how it might work". Might get to work on this today.

willcode, 2018-02-12:
@giladbeeri please give this a try: https://github.com/gnuradio/gnuradio/pull/1612/files

giladbeeri, 2018-02-16:
@willcode ok, I'm leaving comments regarding the code itself in the PR.

willcode, 2018-03-07:
@giladbeeri can this be closed by https://github.com/gnuradio/gnuradio/pull/1612 ?

giladbeeri, 2018-03-07:
as far as I'm concerned - the change works and solved my issue

giladbeeri, 2018-03-07:
Thanks again!

closed 2018-03-07 by giladbeeri