gnuradio/gnuradio #1586: Check origin/licenses on *all* files that don't have a "this file is part of GNU Radio" header, or replace

marcusmueller, 2018-02-06:
#1585 demonstrates this might be a good idea. Things to check (please edit this and add if you find more. I will not close this issue before we know the origin of all files we use):

- [ ] gr::random : box-muller (candidate for source: numerical recipes v3)
- [ ] gr::filter : tap generators (see above | @awalls-cx18 already took care of this)
- [ ]

marcusmueller, 2018-02-06:
Full list of files to check, because they don't contain the "This file is part of GNU Radio" string:

- [ ] cmake/msvc/config.h
- [ ] cmake/msvc/sys/time.h
- [ ] cmake/msvc/unistd.h
- [ ] docs/doxygen/other/shared_ptr_docstub.h
- [ ] docs/doxygen/other/vector_docstub.h
- [ ] gnuradio-runtime/include/gnuradio/thread/thread_group.h
- [ ] gnuradio-runtime/lib/complex_vec_test.cc
- [ ] gnuradio-runtime/lib/high_res_timer.cc
- [ ] gnuradio-runtime/lib/math/sine_table.h
- [ ] gnuradio-runtime/lib/thread/thread_group.cc
- [ ] gr-analog/include/gnuradio/analog/cpfsk_bc.h
- [ ] gr-analog/include/gnuradio/analog/cpm.h
- [ ] gr-analog/lib/cpfsk_bc_impl.cc
- [ ] gr-analog/lib/cpfsk_bc_impl.h
- [ ] gr-analog/lib/cpm.cc
- [ ] gr-channels/lib/sincostable.h
- [ ] gr-digital/examples/ofdm/ofdm_sync.m
- [ ] gr-digital/examples/ofdm/plot_ofdm.m
- [ ] gr-digital/examples/ofdm/ofdm_sync_pn.m
- [ ] gr-digital/include/gnuradio/digital/probe_density_b.h
- [ ] gr-digital/lib/cpmmod_bc_impl.cc
- [ ] gr-digital/lib/probe_density_b_impl.cc
- [ ] gr-digital/lib/probe_density_b_impl.h
- [ ] gr-dtv/include/gnuradio/dtv/atsc_field_sync_mux.h
- [ ] gr-dtv/include/gnuradio/dtv/atsc_interleaver.h
- [ ] gr-dtv/include/gnuradio/dtv/atsc_randomizer.h
- [ ] gr-dtv/include/gnuradio/dtv/atsc_pad.h
- [ ] gr-dtv/include/gnuradio/dtv/atsc_rs_encoder.h
- [ ] gr-dtv/include/gnuradio/dtv/atsc_trellis_encoder.h
- [ ] gr-dtv/include/gnuradio/dtv/catv_reed_solomon_enc_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/catv_transport_framing_enc_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/dvb_config.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbs2_config.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_bit_inner_deinterleaver.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt2_config.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_bit_inner_interleaver.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_config.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_convolutional_deinterleaver.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_convolutional_interleaver.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_demap.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_demod_reference_signals.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_energy_descramble.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_energy_dispersal.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_inner_coder.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_map.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_ofdm_sym_acquisition.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_reed_solomon_dec.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_reed_solomon_enc.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_reference_signals.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_symbol_inner_interleaver.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt_viterbi_decoder.h
- [ ] gr-dtv/include/gnuradio/dtv/catv_config.h
- [ ] gr-dtv/include/gnuradio/dtv/catv_frame_sync_enc_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/catv_randomizer_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/catv_trellis_enc_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/dvb_bbheader_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/dvb_bbscrambler_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/dvb_bch_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/dvb_ldpc_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbs2_interleaver_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbs2_modulator_bc.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbs2_physical_cc.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt2_cellinterleaver_cc.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt2_framemapper_cc.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt2_interleaver_bb.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt2_freqinterleaver_cc.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt2_miso_cc.h
- [ ] gr-dtv/include/gnuradio/dtv/dvbt2_modulator_bc.h
[... 137 lines cut ...]
- [ ] gr-vocoder/lib/codec2/lpc.h
- [ ] gr-vocoder/lib/codec2/lsp.h
- [ ] gr-vocoder/lib/codec2/machdep.h
- [ ] gr-vocoder/lib/codec2/nlp.h
- [ ] gr-vocoder/lib/codec2/os.h
- [ ] gr-vocoder/lib/codec2/phase.h
- [ ] gr-vocoder/lib/codec2/phaseexp.h
- [ ] gr-vocoder/lib/codec2/postfilter.h
- [ ] gr-vocoder/lib/codec2/pilot_coeff.h
- [ ] gr-vocoder/lib/codec2/quantise.h
- [ ] gr-vocoder/lib/codec2/rn.h
- [ ] gr-vocoder/lib/codec2/sine.h
- [ ] gr-vocoder/lib/codec2/test_bits.h
- [ ] gr-vocoder/lib/gsm/gsm.h
- [ ] gr-vocoder/lib/gsm/config.h
- [ ] gr-vocoder/lib/gsm/private.h
- [ ] gr-vocoder/lib/gsm/unproto.h
- [ ] gr-vocoder/lib/gsm/proto.h
- [ ] gr-fcd/examples/c++/fcd_nfm_rx.cc
- [ ] gr-fcd/include/gnuradio/fcd/source_c.h
- [ ] gr-fcd/lib/fcd/fcd.h
- [ ] gr-fcd/lib/fcd/fcdhidcmd.h
- [ ] gr-fcd/lib/source_c_impl.cc
- [ ] gr-fcd/lib/hid/hidapi.h
- [ ] gr-fcd/lib/source_c_impl.h

marcusmueller, 2018-02-06:
Same as above, but for python files:

- [x] [docs/doxygen/doxyxml/generated/__init__.py](https://github.com/gnuradio/gnuradio/blob/master/docs/doxygen/doxyxml/generated/__init__.py)
- [x] [docs/doxygen/doxyxml/generated/compound.py](https://github.com/gnuradio/gnuradio/blob/master/docs/doxygen/doxyxml/generated/compound.py)
- [x] [docs/doxygen/doxyxml/generated/index.py](https://github.com/gnuradio/gnuradio/blob/master/docs/doxygen/doxyxml/generated/index.py)
- [x] [docs/doxygen/doxyxml/generated/indexsuper.py](https://github.com/gnuradio/gnuradio/blob/master/docs/doxygen/doxyxml/generated/indexsuper.py)
- [x] [docs/doxygen/doxyxml/generated/compoundsuper.py](https://github.com/gnuradio/gnuradio/blob/master/docs/doxygen/doxyxml/generated/compoundsuper.py)
- [ ] [docs/sphinx/gnuradio_sphinx.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/gnuradio_sphinx.py)
- [ ] [docs/sphinx/hieroglyph/errors.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/errors.py)
- [ ] [docs/sphinx/hieroglyph/nodes.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/nodes.py)
- [ ] [docs/sphinx/hieroglyph/__init__.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/__init__.py)
- [ ] [docs/sphinx/hieroglyph/test/__init__.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/test/__init__.py)
- [ ] [docs/sphinx/hieroglyph/hieroglyph.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/hieroglyph.py)
- [ ] [docs/sphinx/hieroglyph/test/test_hierglyph.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/test/test_hierglyph.py)
- [ ] [docs/sphinx/hieroglyph/version.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/version.py)
- [ ] [docs/sphinx/hieroglyph/test/test_nodes.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/test/test_nodes.py)
- [ ] [docs/sphinx/hieroglyph/test/test_comments.py](https://github.com/gnuradio/gnuradio/blob/master/docs/sphinx/hieroglyph/test/test_comments.py)
- [ ] [gnuradio-runtime/examples/mp-sched/affinity_set.py](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/examples/mp-sched/affinity_set.py)
- [ ] [gnuradio-runtime/python/gnuradio/gr/tag_utils.py](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/python/gnuradio/gr/tag_utils.py)
- [ ] [gnuradio-runtime/python/gnuradio/gr/gr_threading_24.py](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/python/gnuradio/gr/gr_threading_24.py)
- [ ] [gnuradio-runtime/python/gnuradio/gr/gr_threading_23.py](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/python/gnuradio/gr/gr_threading_23.py)
- [ ] [gnuradio-runtime/python/gnuradio/gr_xmlrunner.py](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/python/gnuradio/gr_xmlrunner.py)
- [ ] [gnuradio-runtime/python/gnuradio/gru/__init__.py](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/python/gnuradio/gru/__init__.py)
- [ ] [gr-blocks/examples/ctrlport/simple_copy_controller.py](https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/examples/ctrlport/simple_copy_controller.py)
- [ ] [gr-blocks/examples/ctrlport/usrp_sink_controller.py](https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/examples/ctrlport/usrp_sink_controller.py)
- [ ] [gr-blocks/examples/ctrlport/usrp_source_controller.py](https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/examples/ctrlport/usrp_source_controller.py)
- [ ] [gr-blocks/python/blocks/qa_hier_block2.py](https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/python/blocks/qa_hier_block2.py)
- [ ] [gr-channels/python/channels/amp_bal.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/amp_bal.py)
- [ ] [gr-channels/python/channels/distortion_2_gen.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/distortion_2_gen.py)
- [ ] [gr-channels/python/channels/conj_fs_iqcorr.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/conj_fs_iqcorr.py)
- [ ] [gr-channels/python/channels/distortion_3_gen.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/distortion_3_gen.py)
- [ ] [gr-channels/python/channels/impairments.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/impairments.py)
- [ ] [gr-channels/python/channels/iqbal_gen.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/iqbal_gen.py)
- [ ] [gr-channels/python/channels/phase_bal.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/phase_bal.py)
- [ ] [gr-channels/python/channels/phase_noise_gen.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/phase_noise_gen.py)
- [ ] [gr-channels/python/channels/quantizer.py](https://github.com/gnuradio/gnuradio/blob/master/gr-channels/python/channels/quantizer.py)
- [ ] [gr-filter/python/filter/gui/icons_rc.py](https://github.com/gnuradio/gnuradio/blob/master/gr-filter/python/filter/gui/icons_rc.py)
- [ ] [gr-filter/python/filter/gui/pyqt_filter_stacked.py](https://github.com/gnuradio/gnuradio/blob/master/gr-filter/python/filter/gui/pyqt_filter_stacked.py)
- [ ] [gr-qtgui/apps/usrp_display_qtgui.py](https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/apps/usrp_display_qtgui.py)
[... cut ...]
- [ ] [grc/blocks/variable_struct.xml.py](https://github.com/gnuradio/gnuradio/blob/master/grc/blocks/variable_struct.xml.py)
- [ ] [grc/core/__init__.py](https://github.com/gnuradio/gnuradio/blob/master/grc/core/__init__.py)
- [ ] [grc/core/utils/complexity.py](https://github.com/gnuradio/gnuradio/blob/master/grc/core/utils/complexity.py)
- [ ] [grc/core/utils/epy_block_io.py](https://github.com/gnuradio/gnuradio/blob/master/grc/core/utils/epy_block_io.py)
- [ ] [grc/gui/__init__.py](https://github.com/gnuradio/gnuradio/blob/master/grc/gui/__init__.py)

guruofquality, 2018-02-06:
>  cmake/msvc/config.h @guruofquality
cmake/msvc/sys/time.h @guruofquality
cmake/msvc/unistd.h @guruofquality do you remember where these files are from?

* cmake/msvc/config.h: I made this one for gnuradio. We can delete all of the `#if _MSC_VER < 1800` at this point. That would make it a lot smaller

* cmake/msvc/sys/time.h: Its from a microsoft forum (like https://social.msdn.microsoft.com/Forums/vstudio/en-US/430449b3-f6dd-4e18-84de-eebd26a8d668/gettimeofday?forum=vcgeneral) Its sort of a go to gettimeofday implementation, a lot of projects use it like https://github.com/LibVNC/libvncserver/blob/master/compat/msvc/sys/time.h But I would be partial to using std::chrono these days :-)

* cmake/msvc/unistd.h: I made this one for gnuradio. I would rather delete it and just conditionally include `<process.h>` in the one or two files that need it. And include guard/remove many of the `<unistd.h>` throughout the code. Being an OS specific header and all.

marcusmueller, 2018-02-06:
For now, I'll say that as a quickfix, we'll just declare `msvc/unistd.h` as "not a problem", as it arguably contains any significant code, and can easily be GPL'ed & assigned to FSF now that we know it's yours; generally, I agree: we should be looking into the `unistd.h` usage globally - if it's really just the same as `process.h` wherever it's included on windows, I wonder whether we're using correctly on *nix. 

time.h would worry me *a lot*, but then: we're not using gettimeofday *anywhere* sensible; thus: we'll just remove that.

marcusmueller, 2018-08-24:
Let's get this over with after we do the next merge and formatting patch. There's anyway a lot of code to be purged in 3.8.

marcusmueller, 2021-01-16:
The goalpost of this has changed with the DCO, technically, but de facto we're as reliant on our tree being clean copyright-wise.

guruofquality, 2021-01-16:
@marcusmueller do you just want a PR to delete these 3 compatibility header files and make sure everything compiles? It shouldnt be a big deal. have a decent visual studio development environment ready to go.

marcusmueller, 2023-12-21:
Remains a valid problem, and might need to be revisited. Much easier now with SPDX!