gnuradio/gnuradio #1588: pm_remez.cc: VOLKize

marcusmueller, 2018-02-06:
Pretty sure that in

	for(i=0; i<k; i++) {
	  D[j] = des[2*band] + i*(des[2*band+1]-des[2*band])/(k-1);
	  W[j] = weight[band];
	  Grid[j] = lowf;
	  lowf += delf;
	  j++;
	}

* `j` should be `i` (why?)
* `D[j] =` nicely vectorizes,
* `W[j] = weight[band]` nicely vectorizes, as band is static to the loop,
* same goes for `Grid[j] = lowf; lowf +=delf`, which is really just a way of saying `Grid[j] = lowf0 + j·delf`

itss1ddhant, 2020-09-15:
I am a beginner in Open Source and wished to take up this issue. Can I get some more details on it as in , if the matter is resolved or not. If not, then where should I start?

marcusmueller, 2021-03-23:
Hi! I don't know whether you're still around, but we'd need you to be at least superficially familiar with the Remez-Exchange filter design algorithm, and with SIMD acceleration. It's a good first issue, if you're a DSP person :)

marcusmueller, 2023-04-10:
Having had a field day with pm_remez lately, I'd say before we vectorize this, there's more about it that needs some TLC.

closed 2023-04-10 by marcusmueller