gnuradio/gnuradio #1591: cleanup gru Python module

marcusmueller, 2018-02-06:
I'm generally not a big fan of "misc" namespaces, and I'm pretty sure that there's a few things in there that no-one uses.

Obviously, low prio.

bravech, 2019-12-26:
Hi, I'm looking to get started with contributing to open source (hopefully joining GNURadio for GSoC 2020!). Is this change still something that's needed? If not, could you please point me in the direction of an issue to get started with.

gateship1, 2020-05-18:
Hello, I am looking for a first issue. I noticed this is still under "To do" in Release 3.9. Can i take a crack at this one?

dkozel, 2020-05-27:
@gateship1 Yes, please do. I'm not clear whether there are only some unused functions that can be deleted, the entire module should be deprecated, or the functions should be moved into the calling functions. You'll have to see what's still in there as this issue is over two years old.

argilo, 2020-10-20:
Re-opening, since gru isn't removed yet.

marcusmueller, 2020-10-29:
With #3896, we have but left in actual use:

- [ ] `daemonize` in the dialtone example. Do we really need to use that? It feels like it was some Python <2.5-era hack
- [ ] `freqz`/`gnuplot_freqz`: used only in commented out code in `fm_emph`, and honestly, what about `scipy.signal`? 
- [ ] `hexshort`: unused
- [ ] `hexint`: used in crc32. `hexint = lambda mask: (mask - 2**32) if (mask >= 2**31) else mask`.
- [ ] `msgq_runner`: single-point usage in `uhd_app.py`. @mbr0wn is moving it to that app itself sensible?

marcusmueller, 2020-10-30:
Thanks, @argilo! Awesome!

PR #3815 (2020-10-20): gru: remove unused files
Partially resolves #1591.

I had a look over `gnuradio.gru` and agree with @marcusmueller's assessment that it's useless and may as well be removed. Here I've started with listmisc.py, os_read_exactly.py, seq_with_cursor.py and socket_stuff.py, all of which are unused within GNU Radio.

The rest of the files have a small number of usages that shouldn't be too hard to remove, but I'll leave them to follow-up pull requests.
closed 2021-01-16 by marcusmueller