gnuradio/gnuradio #1593: cmake/msvc/config.h: delete all of `#if _MSC_VER < 1800`

marcusmueller, 2018-02-06:
due to versioning, these versions are impossible today (thanks @guruofquality for mentioning this in #1586!).

marcusmueller, 2018-02-06:
Oh, and config.h without any MSC_VER < 1800 basically only contains non-threadsafe libc (s)random(), which we need to get rid of, in any case, and ssize_t seems to be declared in other headers, specifically BaseTsd.h. With that, config.h boils down to

 #include <BaseTsd>
 #ifndef __cplusplus
 #  define inline __inline
 #endif
which is much easier on the eye.

marcusmueller, 2018-02-08:
Discussion shows we should support VS prior to 2015, which has `MSC_VER==1900`.

marcusmueller, 2018-02-08:
Update regarding needs: 

`gnuradio-runtime/include/gnuradio/attributes.h` already defines `inline` for MSVC. Very questionable we need this file at all, if all it does is `#include <BaseTsd>` for the three or four places where we use `ssize_t`.

noc0lour, 2018-03-10:
For 3.8 supporting MSC_VER < 1800 could be not worth it since this would mean supporting Visual Studio 2012 which lacks a lot of C++11 features according to this table:
https://msdn.microsoft.com/en-us/library/hh567368.aspx

marcusmueller, 2018-03-22:
Yeah, I'm closing this by virtue of it being superseeded by #1680.

closed 2018-03-22 by marcusmueller