gnuradio/gnuradio #1598: pmt: len_y, len_x: unitialized use

noc0lour, 2018-02-08:
In [pmt.cc#L1133](https://github.com/gnuradio/gnuradio/blob/da0a102e9b076db83fe5a63b3730ac39b588eb92/gnuradio-runtime/lib/pmt/pmt.cc#L1133) the comparison of vectors is most likely false since uninitialized memory is used to compare them.

This breaks the fix in #1597, Initializing with `length()` or not using this helper variable at all would help.

noc0lour, 2018-02-08:
Ok, wait this is even better, maybe not even a bug. The variables are initalized inside `uniform_elements`.

noc0lour, 2018-02-08:
Quick test revealed `len_x` and `len_y` have the correct value after `memcmp` but probably are still undefined during `memcmp`.

marcusmueller, 2018-02-11:
> Quick test revealed len_x and len_y have the correct value after memcmp but probably are still undefined during memcmp.

Exactly.

noc0lour, 2018-02-11:
The commit should have closed this?

closed 2018-02-11 by noc0lour