gnuradio/gnuradio #1608: gr_logger: OOT modules do not generate log messages

mrjacobagilbert, 2018-02-11:
I am in the process of changing std::cout messages in an OOT I maintain to use the GR_LOG API and cannot get any log message output. In-tree logger messages print [as described](https://gnuradio.org/doc/doxygen/page_logger.html) when configured to use stdout and stderr or file based logging, but nothing in the OOT will print. 

It's entirely possible I am not using this properly; I am assuming d_logger is something that GR already has for every block...

Module has no changes to CMake files, no extra includes, just put some statements like this into a C++ block:

`GR_LOG_INFO(d_logger, "******** THIS IS A GR LOG INFO MESSAGE ********");`
`std::cout << "++++++++ THIS IS A STD COUT MESSAGE ++++++++" << std::endl;`

The second line prints but not the first. The module also compiles regardless of what I put for the logger; `d_logger`, `d_logger2`, `asdfasdf` which seems suspect. GR seems to pre-define `d_logger` and `d_debug_logger` but no other loggers, so this is a bit odd to me.

GR was built with liblog4cpp, snip from cmake:

```
...
-- Using install prefix: /opt/gnuradio
--
-- Configuring volk support...
--   Enabling volk support.
--   Override with -DENABLE_VOLK=ON/OFF
--   Override with -DENABLE_INTERNAL_VOLK=ON/OFF
-- ENABLE_GR_LOG set to ON.
-- HAVE_LOG4CPP set to True.
-- LOG4CPP_LIBRARIES set to /usr/lib/liblog4cpp.so.
-- Could NOT find Sphinx (missing:  SPHINX_EXECUTABLE)
--
-- Configuring doxygen support...
--   Dependency DOXYGEN_FOUND = YES
...
```

Posting this here to see if someone else can confirm what is going on - if this is just user error, I'll see if I can clarify the documentation.

mrjacobagilbert, 2018-02-12:
I did a bit more exploring, and I am pretty sure these GR_LOG_* messages are skipped entirely (not compiled?). Putting nonsense in there, or things like divide by zeros compile and run fine (except no log output):

```
d_x = 0;
GR_LOG_INFO(d_logger, boost::format("asdf %1%") % (4/d_x));
GR_LOG_INFO(d_logger, boost::format("asdf %1%") % "string?");
GR_LOG_INFO(d_logger, whateveriputheredoesntmatter@#$234);
```

noc0lour, 2018-02-12:
So, did you try calling first:
`GR_LOG_DECLARE_LOGPTR(d_logger)`
`GR_ASSIGN_LOGPTR(d_logger, myname)`
`GR_LOG_INFO(d_logger, "my_msg")`

To shorten this process you could probably use `GR_INFO(myname, "msg")` macro.

mrjacobagilbert, 2018-02-13:
Ok...figured it out; thanks for the suggestions they helped with the thought process.

It was necessary to do several CMake related things in order to get the right flags everywhere to build with logger/liblog4cpp (before the macros were effectively doing nothing as shown in the logger.h file):

- include the GrMiscUtils CMake module
- call the GR_LOGGER() macro in above module in CMakeLists.txt
- copy over the FindLog4cpp CMake module from GNURadio into the OOT cmake/Modules directory

The macro calls then worked, no other setup necessary. 

Is it desired that the GR_LOG API be supported by default for new modules? If so, I can PR the cmake changes I made, otherwise I can update documentation to reflect necessary steps to use the API.

marcusmueller, 2018-02-13:
as a maintainer: yes, using the logging API in OOTs is desirable. Furthermore, documenting (that can also happen in the shape of e.g. a patch) how to add that to existing OOTs is desirable, too!

noc0lour, 2018-02-13:
Preferably gnuradio should expose some sort of logging API so people only have to link and require `libgnuradio-log` and not liblog4cpp directly.

mrjacobagilbert, 2018-02-13:
I agree with @noc0lour as that was what I had expected to happen, but thats beyond me to make so... For what it's worth, this is also an issue with using ControlPort in OOTs (you need to copy over the FindThrift CMake module or it will be disabled).

While I am not a CMake expert, it seems like OOTs should have access to the GR CMake modules from the GR installation and not need to maintain their own versions of `GrMiscUtils.cmake`, `Find_pkg_.cmake`, etc...

I think a patch is important for existing OOT modules. I'm not sure where to include that though. If someone can point out where it should go, i'll write a PR for documentation updates and the patch. I'll also submit a PR to add the FindThrift and FindLog4Cpp modules to the modtool template.

marcusmueller, 2018-02-13:
> While I am not a CMake expert, it seems like OOTs should have access to the GR CMake modules from the GR installation and not need to maintain their own versions of GrMiscUtils.cmake, Find_pkg_.cmake, etc...

Exactly, we're currently looking towards removing swig/CMake cruft from the gr-newmod template (#1553), so not adding new CMake stuff to the template that we install with GR, anyway, would be a big plus (:switzerland:).

On the other hand, @noc0lour is absolutely right, as we're actually low-key planning to kick log4cpp out of the core, anyway, introducing it as direct dependency to the OOTs would be ... bad. I was actually thinking you've found a way to just have the `GR_LOG*` in OOTs without having to link log4cpp. That would be awesome.

noc0lour, 2018-02-13:
I think for GR 3.7 series we need to document how to use logging in a OOT since there are still some modules out there depending on it. Maybe we get there for 3.8 to have an actually API wrapping log4cpp and ctrlport?

marcusmueller, 2018-02-16:
Closing this issue as #1614 solves the documentational aspect, and #1617 documents the need to have a sane way for OOTs to enable logging easily in release 3.8.

closed 2018-02-16 by marcusmueller

marxwolf, 2020-09-17:
@mrjacobagilbert Hi, I also met this problem in gnuradio v3.7.14 with a OOT module.
Is it 
include(GrMiscUtils.cmake)
GR_LOGGING()

in the top-level CMakeLists.txt to make it work?
Thanks.