gnuradio/gnuradio #1615: zmq.hpp: Not actually part of ZMQ itself, but we only list ZMQ as dependency.

marcusmueller, 2018-02-16:
While some distros ship a zmq.hpp alongside their zeromq devel packages, the zmq.hpp that we actually rely on is not actually part of the C bindings of zmq, but of [cppzmq](https://github.com/zeromq/cppzmq), which is a C++ header-only wrapper for the C API. So, users need both: the cppzmq header `zmq.hpp`, the zmq header `zmq.h`, and of course libzmq. 

We conflate cppzmq with libzmq in our CMake script. I presume that's the case because some distros package that couple kB of header with their libzmq-devel packages, but that's not the case for Fedoroids, or Brew. (I guess it's for Ubuntu)

Thus, we shouldn't be trying to find zmq.hpp by asking pkg-config about zmq (`PKG_CHECK_MODULES(PC_ZEROMQ "libzmq")`), but for cppzmq (which *does* ship an official .pc). However, I don't know whether Ubuntu does ship that .pc. Will have to test
(Raised due to slightly mis-titled #1544)

marcusmueller, 2018-02-16:
Just checked: F27 does have a `cppzmq-devel` package (no `cppzmq` package of its own, as header-only), but that doesn't come with the .pc that's part of the official source. :-1: 

So, in effect, we have no better way of detecting `zmq.hpp` than to look into the default folders and where the `libzmq` .pc points to, which is what we currently do. Suboptimal.

noc0lour, 2018-03-13:
So, what's the proposed fix for 3.7.12 to implement?

marcusmueller, 2018-03-22:
That's an incredibly hard to answer question, @noc0lour. I'm not really keen on pulling deps into the tree, so I'll opt for not doing that and documenting the dependency only.

PR #1708 (2018-03-30): docs: adding zmq, sphinx to build deps, restructure optionals
fixes #1616 
fixes #1615
closed 2018-03-30 by marcusmueller