gnuradio/gnuradio #1617: Properly expose logging API to OOTs

marcusmueller, 2018-02-16:
As #1608 documents, we should have a sane way of adding GR logging to OOTs, as documented by @mrjacobagilbert in 854a692871f51f97384dabcea8435cf87a70ce41 .

jacobgilbert-snl, 2018-04-18:
It was pointed out to me by a colleague that the fix above fails to build on his Fedora system, and the fix noted [here](https://github.com/vamsi765/gr-testinggnuradio/commit/c204fc73a1d040539298f776d2fcd345fcce6394) is necessary. This also still builds on Ubuntu. It's unclear why that fix puts LOG4CPP_INCLUDE_DIRS under link_directories and not include_directories but this is all a bit beyond me.

Properly handling this would be awesome.

mrjacobagilbert, 2018-04-18:
It was pointed out to me by a colleague that the fix above fails to build on his Fedora system, and the fix noted [here](https://github.com/vamsi765/gr-testinggnuradio/commit/c204fc73a1d040539298f776d2fcd345fcce6394) is necessary. This also still builds on Ubuntu. It's unclear why that fix puts LOG4CPP_INCLUDE_DIRS under link_directories and not include_directories but this is all a bit beyond me.

Properly handling this would be awesome.

mormj, 2020-11-18:
Went to take a look at this one, and I can't reproduce (no changes to cmake in the OOT).  It is very possible I am not understanding the issue, but there also were some significant changes to logging along the way here.  

Simple OOT, work function looks like:
```cpp
int foo_impl::general_work(int noutput_items,
                           gr_vector_int& ninput_items,
                           gr_vector_const_void_star& input_items,
                           gr_vector_void_star& output_items)
{
    const input_type* in = reinterpret_cast<const input_type*>(input_items[0]);
    output_type* out = reinterpret_cast<output_type*>(output_items[0]);

    consume_each(noutput_items);

    GR_LOG_INFO(d_logger, "message");
    // Tell runtime system how many output items we produced.
    return noutput_items;
}
```
And the message shows up on my stdout out of the box

@mrjacobagilbert - am I doing this in a way that should expose the issue you were seeing?

jacobgilbert-snl, 2020-11-18:
I tested this in 3.8 and 3.9 and i no longer see the issue. My 3.7 system is not in a good place right now, so I cannot really comment on that, but it does not seem to affect the latest stable or master code.

If someone can confirm that this is fixed in the latest 3.7 that would be good so we can close this out.

marcusmueller, 2021-01-16:
I think with this being in shape for 3.8 and 3.9, we can close this: new OOTs for 3.7 are luckily rare.

closed 2021-01-16 by marcusmueller