gnuradio/gnuradio #1621: `d_found` unused in gr::blocks::peak_detector_XX_impl

marcusmueller, 2018-02-16:
Clean up.

talih0, 2018-02-18:
I can take this on

talih0, 2018-02-18:
@marcusmueller, should I work of the 'next' branch?

marcusmueller, 2018-02-20:
yes, please.

talih0, 2018-02-21:
@marcusmueller 
Before making any changes I ran make test on 'master' and 'next'. 
Unfortunately, the test failed. Is this a problem with my compilation, or are these fails expected? 

Total Test time (real) = 141.71 sec

The following tests FAILED:
         91 - qa_polar_decoder_sc_list (Failed)
         95 - qa_polar_decoder_sc (Failed)
         96 - qa_polar_decoder_sc_systematic (Failed)
         97 - qa_polar_encoder_systematic (Failed)
         98 - qa_polar_encoder (Failed)
        214 - qa_zeromq_sub (Failed)
Errors while running CTest
Makefile:61: recipe for target 'test' failed

noc0lour, 2018-02-21:
I guess you're missing scipy for the `qa_polar` tests and `zeromq_sub` is unfortunately a flaky test.

talih0, 2018-02-21:
thanks, qa_polar test was fixed by installing scipy

closed 2018-08-24 by marcusmueller