gnuradio/gnuradio #1624: Remove unused members

marcusmueller, 2018-02-16:
compiler's getting really verbose about unused private members; run with `-Wunused-private-field` and clean up.

tpalmerstudios, 2018-02-17:
I  don't know much about this project, but I'll start working on it

marcusmueller, 2018-02-17:
@tpalmerstudios that's great! 

Have you built GNU Radio from source before? You'll need a relatively recent Clang or GCC (which version does `gcc -v` or `clang -v` give you?). Fedora 27 might be a good choice to run this on.

In the CMake step, you'd do something like 

```
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo "-DCMAKE_CXX_FLAGS=-Wunused-private-field -Wno-deprecated-declarations" ..
```

to get all the compiler warnings about unused private field. For example, I picked from that long list to get you started: #1621, #1622, #1623 .

In essence, you find these fields in the header, remove them, and `make; make test` to make sure you didn't break anything.

Feel like taking this up?

chaithyagr, 2018-02-21:
I'll work on this.. will get back with any issues

closed 2018-08-24 by marcusmueller