gnuradio/gnuradio #1625: snd_pcm_sw_params: Invalid argument accessing hw:1

regulatre, 2018-02-17:
Flow graphs accessing the audio source device when the device is hw:1 and the flowgraph is running on a Raspberry Pi 2b (possibly others) experience problems opening the input device. 

Devices in question seem all use an external USB audio adapter which shows up as hw:1 (since during boot the onboard sound system gets hw:0). The on-board Raspberry Pi audio system does NOT have an input source so hw:0 does not exist. Only hw:1 exists as an input. This seems to be the key to this issue - (the fact that hw:0 input is absent)

```
self.audio_source_0 = audio.source(samp_rate, 'hw:1', True)
``` 

the error (I enabled verbose logging using ~/.gnuradio/config.conf): 
```
gr::log :INFO: audio source - Audio source arch: alsa
PCM name: hw:1
Access types:
    MMAP_INTERLEAVED     YES
    MMAP_NONINTERLEAVED  NO
    MMAP_COMPLEX         NO
    RW_INTERLEAVED       YES
    RW_NONINTERLEAVED    NO
Formats:
    S16_LE               YES
Number of channels
    min channels: 1
    max channels: 1
    1 channels	YES
Sample Rates:
    min rate:   44100 (dir = 0)
    max rate:   48000 (dir = 0)
      8000  NO
     16000  NO
     22050  NO
     32000  NO
     44100  YES
     48000  YES
     96000  NO
    192000  NO
audio_alsa_source[hw:1]: using S16_LE
gr::log :INFO: audio source - Audio sink arch: alsa
PCM name: hw:1
Access types:
    MMAP_INTERLEAVED     YES
    MMAP_NONINTERLEAVED  NO
    MMAP_COMPLEX         NO
    RW_INTERLEAVED       YES
    RW_NONINTERLEAVED    NO
Formats:
    S16_LE               YES
Number of channels
    min channels: 2
    max channels: 2
    2 channels	YES
Sample Rates:
    min rate:   44100 (dir = 0)
    max rate:   48000 (dir = 0)
      8000  NO
     16000  NO
     22050  NO
     32000  NO
     44100  YES
     48000  YES
     96000  NO
    192000  NO
audio_alsa_sink[hw:1]: using S16_LE
Raw input RMS (x1000): 0.0
gr::log :DEBUG: audio_alsa_source0 - [hw:1]: sample resolution = 16 bits
gr::log :ERROR: audio_alsa_sink0 - [hw:1]: snd_pcm_sw_params: Invalid argument
Traceback (most recent call last):
  File "./acpqheadphones.py", line 146, in <module>
    main()
  File "./acpqheadphones.py", line 131, in main
    tb.start()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 4879, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: audio_alsa_sink
```

List of input devices 

```
$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [C-Media USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

regulatre, 2018-02-17:
In the past I had limited success by using a gnuradio config that specified hw:1 as the default audio device, but today that trick isn't working, which precipitated me to open this ticket. Hopefully I provided enough info to go on. Thank you!

marcusmueller, 2018-02-17:
Hi @regulatre! I'm not sure you're actually submitting a bug report for GNU Radio – feels like "the default value of `hw:0` doesn't work", but to be frank here: it's a rather strange config on your OS's side to have a hw:1 but not a hw:0 (there's nothing saying that USB soundcards can't have any number, these numbers are picked by ALSA and your OS's configuration); anyway, it's just a default value, and using a different value would break it for all others; not quite sure what to recommend to solve the situation here.

I really think you should be documenting this in the [Wiki](https://wiki.gnuradio.org/index.php/ALSAPulseAudio) instead of an issue here: That would make it easy for Pi users to find the solution.

Would you agree, or could you elaborate on why this is a GNU Radio bug?

regulatre, 2018-02-18:
If I knew of a way to push past the error I would tend to agree, however at this point I haven't been able to run the flow graph after thousands of tries (put it in a loop to see if it would eventually be successful) also tried rebooting the pi, and tweaking the config.conf (which worked occasionally in the past). 

I'm afraid I don't know of a work-around. 

I'm able to happily record audio using arecord from the commandline, however as far as I can tell, gnuradio itself is having trouble (Gnuradio seems to be trying to open hw:0 even though that device is not specified anywhere, and in doing so it crashes itself with the fatal error specified).

regulatre, 2018-02-18:
I just tried manually setting the alsa device name using udev rules and it worked once, then subsequent attempts to run the flowgraph resulted in the same error again. 

Error message: 
```
gr::log :ERROR: audio_alsa_sink0 - [hw:VOLTS_A]: snd_pcm_sw_params: Invalid argument
```

udev rules (assign a unique hw:XXX device name to sound devices based on the Raspberry Pi USB port they are connected to): 

```
SUBSYSTEM!="sound", GOTO="my_usb_audio_end"
ACTION!="add", GOTO="my_usb_audio_end"

DEVPATH=="/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/sound/card?", ATTR{id}="VOLTS_A"
DEVPATH=="/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0/sound/card?", ATTR{id}="VOLTS_B"
DEVPATH=="/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/sound/card?", ATTR{id}="CUR_A"
DEVPATH=="/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/sound/card?", ATTR{id}="CUR_B"

LABEL="my_usb_audio_end"
```

Source: https://github.com/regulatre/gr-powerquality/blob/master/examples/udev-rules/70-alsa-permanent.rules 

Verbose error: 

```
$ python ./acpqheadphones.py 
gr::log :INFO: audio source - Audio source arch: alsa
PCM name: hw:VOLTS_A
Access types:
    MMAP_INTERLEAVED     YES
    MMAP_NONINTERLEAVED  NO
    MMAP_COMPLEX         NO
    RW_INTERLEAVED       YES
    RW_NONINTERLEAVED    NO
Formats:
    S16_LE               YES
Number of channels
    min channels: 1
    max channels: 1
    1 channels	YES
Sample Rates:
    min rate:   44100 (dir = 0)
    max rate:   48000 (dir = 0)
      8000  NO
     16000  NO
     22050  NO
     32000  NO
     44100  YES
     48000  YES
     96000  NO
    192000  NO
audio_alsa_source[hw:VOLTS_A]: using S16_LE
gr::log :INFO: audio source - Audio sink arch: alsa
PCM name: hw:VOLTS_A
Access types:
    MMAP_INTERLEAVED     YES
    MMAP_NONINTERLEAVED  NO
    MMAP_COMPLEX         NO
    RW_INTERLEAVED       YES
    RW_NONINTERLEAVED    NO
Formats:
    S16_LE               YES
Number of channels
    min channels: 2
    max channels: 2
    2 channels	YES
Sample Rates:
    min rate:   44100 (dir = 0)
    max rate:   48000 (dir = 0)
      8000  NO
     16000  NO
     22050  NO
     32000  NO
     44100  YES
     48000  YES
     96000  NO
    192000  NO
audio_alsa_sink[hw:VOLTS_A]: using S16_LE
Raw input RMS (x1000): 0.0
gr::log :DEBUG: audio_alsa_source0 - [hw:VOLTS_A]: sample resolution = 16 bits
gr::log :ERROR: audio_alsa_sink0 - [hw:VOLTS_A]: snd_pcm_sw_params: Invalid argument
Traceback (most recent call last):
  File "./acpqheadphones.py", line 146, in <module>
    main()
  File "./acpqheadphones.py", line 131, in main
    tb.start()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 4879, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: audio_alsa_sink
```

regulatre, 2018-02-19:
Update: I boiled down the app to the simplest form so I could pinpoint what was happening and I found a completely different issue was at play. 

I have a probe connected to the audio input block and a thread that starts immediately trying to check the level() of that probe. As soon as I disabled that thread and re-ran the flowgraph I found the error has disappeared. 

It seems the probe was trying to poll the value of the input *before* it was ready or fully initialized. 

A bit of a race condition was at play - to address the issue, I created a new class function called `startThreads()`, which contains my thread `.start()` call. I run startThreads from `main()` after `tb.start()`, and this seems to be the winning recipe. 

Today I learned: don't let a probe thread start running before the flowgraph gets start()'d. 

Closing issue.

closed 2018-02-19 by regulatre

marcusmueller, 2018-02-19:
Thanks! That is actually a valuable insight that I wished we transported to users of probes somewhere. Keep up the good work!

noc0lour, 2018-02-19:
Yeah, I've had some bad experiences with function probes, they"re usually not the way to go...

regulatre, 2019-01-17:
This behavior is rearing its ugly head again. It's a really hard one to pin down because it pops up out of nowhere and goes away without having changed the code, OS, or hardware. The behavior is that while developing a flowgraph that uses audio sink and source (same device for both) while starting up, the app crashes with "snd_pcm_sw_params: Invalid argument".  It seem like a timing or memory leak issue. It looks like Marcus ran into it some time ago (https://lists.gnu.org/archive/html/discuss-gnuradio/2015-08/msg00200.html) and I'm afraid I'm not handy with gdb or other memory debugging applications. 

I should mention that this development is all being carried out on Raspberry Pi devices. Being that their clock speed is slower may be a factor related to why this doesn't show up on faster systems. As the flowgraph is starting up is when the error occurs. 

Looking for any suggestions that I can apply to zero in on the issue. 

Thanks