gnuradio/gnuradio #1626: hier_block2 exposes set_msg_handler in C++ but not in Python

beaugunderson, 2018-02-18:
`set_msg_handler` is exposed on `hier_block2` in C++ but not in Python (I'd like to use it in a Python block).

If someone can point me in the right direction I'm happy to work on a PR. :)

willcode, 2018-02-18:
C++ hier_block2 subclasses basic_block, where the Python version subclasses object. You could experiment with making Python's version subclass basic_block (easy enough to try) and see what happens. Are you looking to be able to dynamically route messages to different blocks, or sniff messages, or something like that?

beaugunderson, 2018-02-18:
@willcode ah, I see--I did find a workaround for what I wanted to do though.

I had a `hier_block2` "a" that contains a block "b" with a `b_out` port, and I wanted users of "a" to be able to receive messages from the `b_out` port.

This did the trick without needing to use a message handler:

```python
self.message_port_register_hier_out('a_out')
self.msg_connect((self.b, 'b_out'), (self, 'a_out'))
```

It was not immediately obvious that connecting one out port to another out port would work, but it did!

noc0lour, 2018-02-18:
You can document this in the wiki :)

marcusmueller, 2018-02-19:
I'm considering closing this issue now that you have an appropriate workaround. I'd generally be careful with having actual message handlers in hier blocks - these actually don't come with a block executor, so who would run that message handler?

beaugunderson, 2018-02-19:
@marcusmueller yes, I'll close--and that is also useful information, and wasn't immediately apparent from the hier_block2 docs (though I did see a mailing list message about it from 2013, but wondered if it was out of date) :)

closed 2018-02-19 by beaugunderson