gnuradio/gnuradio #1629: QT GUI blocks do not replicate all functionality of WX blocks

dkozel, 2018-02-21:
This issue tracks the features missing in the QT blocks compared to WX.

## Missing Features

- [ ] XY plotting
- [ ] Stripchart time sink
- [ ] Histogram chart

marcusmueller, 2018-02-21:
This would by the way make great GSoC features :) can you check whether this might be an amendment to an existing GSoC idea or a new one? Note that listing it as GSoC idea wouldn't mean we discourage non-GSoC contributions!

dcajacob, 2018-02-22:
Might want to pull in my latest issue since this seems to be a good umbrella for it: https://github.com/gnuradio/gnuradio/issues/1631

dkozel, 2018-02-26:
Also relevant, putting time and effort into Qwt based blocks isn't very useful if we'll be moving to the pyqtgraph library.
https://github.com/gnuradio/gnuradio/issues/1634

noc0lour, 2018-02-26:
The problem is PyQwt, Qwt is not affected, If someone dares to rewrite examples and apps in C++ Qwt is on the table.

marcusmueller, 2018-03-02:
@noc0lour I remember "back in my days", that QWT was a dependency nightmare, has that gotten better? I'm afraid of QWT5/6 on QT4/5 fun all the way to 3.9.

noc0lour, 2018-03-03:
Dependency graph doesn't look too bad? 
```
 * dependency graph for x11-libs/qwt-6.1.3-r1
 `--  x11-libs/qwt-6.1.3-r1  amd64 
   `--  virtual/opengl-7.0-r1  (virtual/opengl) amd64 
   `--  dev-qt/qtcore-4.8.7-r4  (dev-qt/qtcore) amd64 
   `--  dev-qt/qtgui-4.8.7  (dev-qt/qtgui) amd64 
   `--  dev-qt/designer-4.8.7  (dev-qt/designer) amd64 
   `--  dev-qt/qtopengl-4.8.7  (dev-qt/qtopengl) amd64 
   `--  dev-qt/qtsvg-4.8.7  (dev-qt/qtsvg) amd64 
   `--  dev-qt/qtconcurrent-5.9.4  (dev-qt/qtconcurrent) ~amd64 
   `--  dev-qt/qtcore-5.9.4-r2  (dev-qt/qtcore) ~amd64 
   `--  dev-qt/qtgui-5.9.4-r3  (dev-qt/qtgui) ~amd64 
   `--  dev-qt/qtprintsupport-5.9.4  (dev-qt/qtprintsupport) ~amd64 
   `--  dev-qt/qtwidgets-5.9.4-r1  (dev-qt/qtwidgets) ~amd64 
   `--  dev-qt/designer-5.9.4  (dev-qt/designer) ~amd64 
   `--  dev-qt/qtopengl-5.9.4  (dev-qt/qtopengl) ~amd64 
   `--  dev-qt/qtsvg-5.9.4-r1  (dev-qt/qtsvg) ~amd64 
```

marcusmueller, 2018-03-03:
I'd agree, that looks fine enough.

jdemel, 2018-03-11:
Is XY plotting really missing from Qt GUI? This is essentially a constellation plot.
As a quick test, I put together a small flowgraph for [Oscillofun](https://www.youtube.com/watch?v=RkJdADVy_Mg). It consists of a 2-channel wav source, connected to a 2-c audio sink. The plotting is either done with `WX Gui Scope Sink` in XY mode, or with an additional `float to complex` block and a Qt Constellation Sink. 
It does not look as amazing as in the linked video, but it works. Altough, this would be a great addition if it could look as amazing as in the video. This might draw interest to GR.